commaai/openpilot #2696: Implement a Low Voltage Disconnect in next generation hardware (stranded/dead battery fix)

zorrobyte, 2020-12-04:
**What are we trying to solve?**
https://github.com/commaai/openpilot/issues/2677

An edge case exists where OP software can boot to a bad state where the phone's backlight is on while the screen is black. I've personally had at least two events where I was stranded due to this, one while camping (and it only took 6-8 hours for my Prius's tiny battery to drain).

Also, lead-acid batteries will reduce their usable lifespan if drained past 50% depth of discharge by up to half, and especially considering starting batteries are even more vulnerable. At 10~ total discharges due to issues with OP over the last year, my Prius's battery is now in need of (costly) replacement.

Consider at even 173mA, a brand new Prius Prime battery will be at 50% charge after just 5.5 days, something the battery is not designed for (more calculations in above issue)
[image]

**Ideas to resolve**
Implement a LVD in comma power to switch off power at 11.7 volts; for example, or even further upstream in Uno. Next generation hardware should have some sort of native implementation, especially for consumer release (software will always eventually crash).

**Immediate fixes**
I've installed one of these https://www.amazon.com/PriorityStart-12-Volt-Pro-SP-Automatic-Battery-Protector/dp/B0030A10V0/ref=sr_1_3?dchild=1&keywords=Priority+Start+12-Volt&qid=1606925871&sr=8-3 - same concept (no, I'm not affiliated with them in any way)

AkshatGiri, 2020-12-06:
Same issue here. My Prius battery died twice because of this. The first time I didn't think comma could be the cause but the second time it was clear.

Flekoun, 2020-12-07:
After 8.0 my Toyota Corolla 2.0 TS (2020) experienced complete battery drain 3 times in a row! Fix this pls. For now I am unplugging the comma 2 until this is fixed. Add an option to completely auto turn off the comma 2 after the car is powered off.

VirtuallyChris, 2020-12-08:
It should have been fixed via https://github.com/commaai/panda/commit/9f5f437c90e604623ffe2813704d0943d6097a87 and a hotfix to 0.8. What zorrobyte is talking about is if the device doesn't build, and fails to run the logic we added to prevent battery drain.

carasen12, 2020-12-15:
Se puede usar en coche hibrido?

robbederks, 2020-12-18:
Thanks for the suggestion. The problem isn't very easy to solve with voltage measurements (too much noise, voltage drop due to current, not a very good indication of state-of-charge, ...), and that's why we now have the power integration built into openpilot. This basically is coulomb-counting both the discharge due to the device's power usage, and also estimating the SOC of the car battery depending on how long it has been on-road for example.
Designing this kind of system is not trivial in hardware, but we'll keep it in mind for potential future hardware. We agree that the experience of a dead car battery is terrible, and must be avoided wherever possible!

closed 2020-12-18 by robbederks

zorrobyte, 2022-01-27:
Bumping as I mentioned this in EPHOT for C3, maybe it can make it in for C3 Slim, newer HW cc @geohot @robbederks -- Consider re-opening this issue.

Discord links to users complaining about dead batteries (including Comma 3):
https://discordapp.com/channels/469524606043160576/871838269405556736/936034635249512498
https://discordapp.com/channels/469524606043160576/774341890089680946/932113360344989697
https://discordapp.com/channels/469524606043160576/524611978208215070/906647291572650014
https://discordapp.com/channels/469524606043160576/469532500335656972/857720814894710795
https://discordapp.com/channels/469524606043160576/525718620517564446/836980557216612382
https://discordapp.com/channels/469524606043160576/538741329799413760/819251996354805801
https://discordapp.com/channels/469524606043160576/524593232290775051/781001330155978752
https://discordapp.com/channels/469524606043160576/525718620517564446/774108398890778634

Some sort of hardware LVD would be helpful built into the Comma Power/C3. Maybe something like this? https://www.mouser.com/datasheet/2/268/MIC2779_Voltage_Monitor_with_Adjustable_Hysteresis-1891106.pdf