commaai/openpilot #2699: Camera frozen after a few minutes of running (VIDEO)

SamuelSandoval, 2020-12-05:
**Describe the bug**

https://www.youtube.com/watch?v=WJ2aGuhEZ94

After getting my drink from Starbucks, I did not engaged the OpenPilot because I was not near the highway.... then noticed the Comma 2 got frozen as seen in the video then rebooted itself. The comma 2 was freshly installed with OpenPilot URL, not the test/staging URL.

2020 Toyota Corolla Hatchback, I have the C2 for about 8-9 months now.

**How to reproduce or log data**

Unable to reproduce the bug. 

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: Comma 2
 - Dongle ID: 92cab8d65e6531b8
 - Route: https://my.comma.ai/92cab8d65e6531b8/1607127877696/1607128179755 (FROZEN HAPPEN) and https://my.comma.ai/92cab8d65e6531b8/1607128291370/1607129253367
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: 8.0 stock version
 - Car make/model: 2020 Toyota Corolla Hatchback

**Additional context**

VIDEO: https://www.youtube.com/watch?v=WJ2aGuhEZ94

SamuelSandoval, 2020-12-08:
Happened again,

This time, Comma froze while OP was engaged

https://www.youtube.com/watch?v=NiR7wGmLjvA

pd0wm, 2020-12-10:
Does it restart after a while or just stay frozen?

Regardless, it sounds like a hardware issue specific to your device if it happens this often. Please contact support@comma.ai and they can arrange a replacement.

SamuelSandoval, 2020-12-11:
Sent. It reboots when it freezes for a few seconds on engaged/disengaged.

pd0wm, 2020-12-15:
I assume this will be solved with a replacement device, so closing this issue.

closed 2020-12-15 by pd0wm