commaai/openpilot #2706: Comma2 does not startup after restarting car

wongjunjin, 2020-12-06:
**Describe the bug**

I noticed this only on v0.8.0. My comma2 does not startup after restarting my car. I would need to unplug the usb-c cable and replug it back into my comma2 for it to start again. This happens only when I leave my car for a certain amount of time (unsure how long) but my C2 screen will go dark. When I restart my car in this state, my C2 will not startup. 

I tried turning off my car, waiting for the C2 to go dark and immediately turn on my car. At this moment, there is no issue, my C2 starts up. So I'm not sure how long I would need to leave my car. Both all of my experience is leaving my car for more than an hour. 

**How to reproduce or log data**

1. Power up C2 by plugging to the car.
2. Use C2 as normal.
3. End trip and turn off the engine. 
4. Leave the car for an amount of time (at least an hour).
5. Return to car and start the engine.
6. C2 will not start. Will need to unplug and replug the USB-C power cable. 

**Expected behavior**

C2 starts up when the car engine starts. This is expected since v 0.7.10.

**Device/Version information (please complete the following information):**
 - Device: comma 2
 - Dongle ID: 7bc774bc8e17fd8a
 - Route: None because it is not captured
 - Timestamp: None because it is not captured
 - Version: v.8.0
 - Car make/model: Toyota Corolla 2019

pd0wm, 2020-12-07:
We looked into the logs, and the only thing we can find is that thermald no longer reports every minute, but only once ever ~2.5 minute. This indicates that boardd crashed. However, I don't see any crashlogs for your device, or any uptick in boardd crashes in general.

1. Do you have any crashlogs in /data/tombstones related to boardd?
2. When this happens, can you SSH in and confirm that boardd crashed?

wongjunjin, 2020-12-09:
Hi thanks for this. 

1. Looking into /data/tombstones, I only have dsps, lpass, modem, and wcnss. Is there anything I need to search for boardd?
2. How do I confirm that the boardd crashed?

pd0wm, 2020-12-09:
1. There are files names `tombstone_xx`. Does any of them mention boardd?
2. You would see pandad/boardd being red in the manager output.

wongjunjin, 2020-12-10:
For item 1, here's what it looks like: 

_root@localhost:/data/tombstones$ dir
dsps  lpass  modem  wcnss
root@localhost:/data/tombstones$_ 

There's nothing on tombstone_xx. 

For item 2: I don't see anything that's being red when I SSH in. 
[image]

I just tried with another drive, and it is still the same. After leaving my car for sometime, it still wouldn't restart when I start my engine. I would need to unplug the USB and replug it back. 

This did not happen at 0.7.10. Should I reinstall openpilot?

pd0wm, 2020-12-10:
Then there are no crashlogs. You could try reverting to 0.7.10 and see if that fixes it. But would still like to know what's causing it, there might be more people with the same problem.

Can you run `lsusb` when it's in a state where it won't start? Let's see if the panda is still connected to the USB. That should help us narrow down the problem.

wongjunjin, 2020-12-10:
Sure. 

Sorry a bit of a non-tech guy here. When you request to run lsusb, how do I do that? SSH into my C2 at the state when it's not starting?

pd0wm, 2020-12-11:
Yes. That is correct.

wongjunjin, 2020-12-11:
Ok will try it tomorrow. Btw, i just tested, after restarting my car (and while my C2 is still in the "off" state), I unplugged my panda and replug it back in. No difference. Still dead.

But will try with the SSH lsusb tomorrow.

pd0wm, 2020-12-11:
How did you unplug the panda in a C2? It's inside the device.

pd0wm, 2020-12-14:
The macbook charging cable is not compatible, it doesn't have all the data lines. Try flipping the USB-C connector on both sides, maybe one side of the connector is broken. Only USB 3.1 gen 2 cables will work properly.

Think I'm understanding the off state correctly. I assumed it was just sitting on the home screen and not switching to the camera view when the car turned on. It's just completely off after a couple of hours?

wongjunjin, 2020-12-15:
So another update:

I flipped it so much that it suddenly started working. But it's bent so badly, i'm buying new cables. 

So it's working fine now, starting up like normal. It even starts when i restart my car too. I've downgraded to version 0.7.10 so i'm not sure if that is the reason. 

Will try more again when I get my new cables. Will upgrade it back to 0.8.0 then and see if the cable makes a difference. 

For the "off state", it's not even on the home screen. It's just black. Totally not switching on. 

The home screen you are mentioning was a scenario i was facing when i was flipping the cables around, and made it worst (another issue that doesnt happen already) - so let's ignore this part.

pd0wm, 2020-12-15:
Keep me updated!

If you keep having issues it might just be a hardware issue. In that case just contact support@comma.ai and we can arrange a replacement device.

wongjunjin, 2020-12-15:
Alright! 

Just tested again with my “faulty” cable on 0.7.10. The “off state” came back. Restarting my car keeps my C2 off. 

I’m suspecting (and hoping) it’s the cable. And not other hardware issue. 

Waiting for the new cable to arrive and will update you. It’s coming from China. Will need to wait for a bit.

wongjunjin, 2020-12-21:
Finally received my new cable. No difference. I'm still getting the "off state" when I restart my car. 

What I found out during the "off state" is that I can switch on my C2, when I hold down the power button. It'll vibrate, and powerup. I never need to do that last time. 

Although with this new method, it's much more bearable to just turn it on, _rather than to unplug and replug the usb-c back._ However, would this be a hardware issue?

VirtuallyChris, 2021-01-27:
Does this still happen on 0.8.1? If so please reach out to support@comma.ai we can replace the device for you.

closed 2021-01-27 by VirtuallyChris