commaai/openpilot #2726: adaptive cruise will be forcefully disengaged after 3 minutes of standstill on Toyota Hybrids

cydia2020, 2020-12-09:
**Describe the bug**
With openpilot's current standstill, ACC will be forcefully disengaged by the car after exactly 3 minutes if the front car doesn't move. In this situation, the car will be shifted to park. This is a potential problem for comma pedal users as comma pedal users tend to not move the cruise control lever when the car is stopped, if the front car doesn't move due to heavy traffic, the car will be shifted to P

**How to reproduce or log data**
Stop behind a completely stationary car in a Toyota hybrid vehicle, after 3 minutes with no movements, ACC will be forcefully disengaged and the car shifted to P. 

**Expected behavior**
Enter the "Waiting" mode (seems to be related to `CRUISE_CONTROL_STATE` 11 in `PCM_CRUISE_SM`) instead of the current  "operate cruise lever or accelerator pedal to resume" mode.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 399b1cca765e80b4
 - Route: 399b1cca765e80b4|2020-12-09--11-54-59
 - Timestamp: the final disengagement in this route
 - Version: 0.8.1
 - Car make/model: Toyota Prius 2018

**Additional context**
With openpilot's standstill logic, the car's dash will immediately display a "operate cruise lever or accelerator pedal to resume" message upon entering standstill, while the stock system will display the "waiting" message and only display "operate cruise lever or accelerator pedal to resume" after the front car has moved, which doesn't seem to trigger this lock-out.

cydia2020, 2020-12-09:
> TIL that if you sit with cruise control engaged in stand still traffic, eventually a 2017 Prius Prime will put itself in park. This causes OP to freak out and alarm.

https://discord.com/channels/469524606043160576/524327905937850394/761394246493929482
This has been documented before on the discord server.

briskspirit, 2020-12-10:
Waiting mode is 7, mode 11 is a 3 seconds timer after which PCM enters mode 7. 
Same will happen if you will wait too long in the mode 7 (Tested on Corolla Hybrid 2020).
What I've discovered for TSS2 models: OP don't need to send standstill request. Could be true for all Toyotas, need to check.
If you would like to participate in the tests - DM me on the discord.

cydia2020, 2020-12-10:
> Waiting mode is 7, mode 11 is a 3 seconds timer after which PCM enters mode 7.
> Same will happen if you will wait too long in the mode 7 (Tested on Corolla Hybrid 2020).
> What I've discovered for TSS2 [...]

Right, I've timed it, same will happen in waiting mode on stock.

However, op will not be able to restart with the SnG hack (at least on TSS-P) applied if the car doesn't roll forward.

briskspirit, 2020-12-10:
Could you do one test? carcontroller.py try to comment out lines 92, 93 , reboot and try. Will it still go into P in 3 minutes?

cydia2020, 2020-12-10:
> Could you do one test? carcontroller.py try to comment out lines 92, 93 , reboot and try. Will it still go into P in 3 minutes?

No, but this is more or less expected since the car doesn't go into standstill

adeebshihadeh, 2022-04-27:
I'm not sure why I understand this is a problem. It sounds like this is all handled gracefully within the car, cruise is disabled and parking brake is engaged. Since openpilot's state is tied to the car's cruise state, it should handle this gracefully as well.

cydia2020, 2022-05-02:
I feel like there should be an alert, as the user can prevent this from happening, with the current implementation of standstill, the user can press the resume button to get the car out of standstill mode.