commaai/openpilot #2728: CAN Error: Check Connections (Kona 2020 SE)

raunaqchamdia, 2020-12-09:
**Describe the bug**

Whenever I start the car up, the screen just say's "CAN Error: Check Connections" after openpilot starts. 

**How to reproduce or log data**

At the car start

**Expected behavior**

I'd expect the comma to behave as it should, taking over the controls when cruise control is activated.

Device/Version information (please complete the following information):

Device: [comma two]
Dongle ID: [df8f689e0634bc3a]
Route: [df8f689e0634bc3a|2020-12-09--10-57-15]
Timestamp: [At initialization/ startup]
Version: [0.8]
Car make/model: [Hyundai Kona 2020 SE]

**Additional context**
I have added fingerprinting 2.0 for the car in values.py as per the instructions, the device switches from dashcam to openpilot mode after the change. The CAN Error has been there since i first plugged the device in, before I modified the values.py. I rechecked all the device connections to make them secure but i still see this error.

pd0wm, 2020-12-10:
Does your car have SCC? This is a requirement for openpilot to work. Not sure which trim levels on the Kona have SCC.

The full logs for your route are not uploaded yet so can't check. Please connect to wifi so they can upload.

raunaqchamdia, 2020-12-10:
Hi Willem,

Yes this Hyundai Kona 2020 includes Forward Collision Avoidance Assist (FCA/SCC/ACC) and LKA systems. I did notice that I did not find the fwRadar ECU firmware while fingerprinting 2.0, do I need to activate cruise control to get this value? I have attached my cars feature sheet for reference if required. I had actually specifically purchased this model since I wanted a comma two :)

About connecting my comma two to wifi, will a cellphone hotspot be enough? I don’t really know how I could get wifi in my parking lot.

Regards,
Raunaq Chamdia

pd0wm, 2020-12-11:
Forward Collision Avoidance Assist doesn't mean that it has adaptive cruise control. The lower trim levels implement FCA using the camera only.

I managed to request an upload of a short route from your devices over cellular, and don't see any SCC related messages. (`df8f689e0634bc3a|2020-12-09--17-14-00`). Together with the missing firmware I can only conclude that your car does not have SCC.

(Sending attachements to github issues over email doesn't work. You'd have to upload it on the website).

raunaqchamdia, 2020-12-11:
Got it, that's unfortunate since FCA also has a high-res camera on the front I had thought it would work. Thanks for the assistance.

closed 2020-12-11 by raunaqchamdia