commaai/openpilot #2731: Got pre collision warning with AEB activation, OP haven't seen the stopped semi?

briskspirit, 2020-12-10:
Was going off ramp with high speed, on a stop light there was a stopped semi truck. TSS2 issued AEB warning and applied brakes, OP haven't seen stopped lead car? 

Not bragging about AEB at all, creating this bug report to check if there is a bug with lead detection or if model can't see semis.

P.S.: had problems with semis in front before, sometimes they are transparent and sometimes there's a feeling that distance to lead is calculated to its rear axle(radar error?)

Route: f66032c2b5aa18ac|2020-12-09--09-38-44
Timestamp: 1923 sec
Version: 0.8.1 master-ci

abdulrahman004, 2020-12-10:
Going to add a similar issue that occurred to me. A car entered sideways and merge into my lane. OP didnt detect it till very late. My car's stock AEB did fine and beeped at me (not sure if it applied brakes)

Route:59f457ce47259dd2|2020-12-10--02-48-45--3
Timestamp: 18:51:45
Version: 0.8.0 - release2

gif https://imgur.com/a/Zkm9hVp

pd0wm, 2020-12-10:
It was commanding brakes, but seems like the car took quite a while to respond. Due to the slow response it also required more braking than the 3 m/s^2 allowed by the safety limits.

[image]

pd0wm, 2020-12-10:
@abdulrahman004 The model only detected the car once it was fully in your lane. Hopefully end-to-end longitudinal will be better at dealing with these situations.

briskspirit, 2020-12-10:
@pd0wm Could be that it was reacting slow because it was me just setting cruise speed lower and lower while going off ramp?
Status value 10:

[image]

HaraldSchafer, 2022-06-07:
https://github.com/commaai/openpilot/issues/24746

haraschax, 2023-05-23:
There is no known way to make TSS2 more responsive

closed 2023-05-23 by haraschax