commaai/openpilot #2736: There was an error - failed to download recovery

DEEgao888, 2020-12-10:
At the very last stage, the comma 2 device request to update the NEOS and keeping showing the message "There was an error - failed to download recovery."

Does anyone know how to fix it?

VirtuallyChris, 2020-12-10:
Can you reflash the NEOS operating system onto the device? A guide on how to do this is here for Windows, Mac, or Linux. https://github.com/commaai/eon-neos#restoring-on-linuxos-x

adeebshihadeh, 2020-12-10:
This only happens when downgrading to an openpilot version which uses NEOS 14. If you'd like to downgrade, you can flash your C2 using the appropriate commit in [eon-neos](https://github.com/commaai/eon-neos) for your openpilot version, otherwise stick with the current release version.

closed 2020-12-10 by adeebshihadeh

DEEgao888, 2020-12-10:
I did this last night using Mac but it require a password from Mac OS terminal to finish the installation. Can you let me know the password?

Di Gao, Ph.D.

DEEgao888, 2020-12-10:
Hi Adeeb,

Thanks for letting me know.

Can you let me know how to downgrade NEOS 14? on top of the instructions
below? I believe the code below is for the newest version? right?
Restoring on Linux/OS X

   1. Connect your comma two (via a USB-C to USB-A cable) or EON (via a
   USB-mini-B to USB-A cable) to your computer
   2. Open a terminal
   3. Clone this repo git clone https://github.com/commaai/eon-neos.git,
   then cd into the cloned repo cd eon-neos
   4. Run ./download.py
   5. Put your device into fastboot mode by turning off your device, then
   holding volume down + power (comma two, EON Gold), or volume up + power
   (EON).
   6. Run ./flash.sh DO NOT DISCONNECT THE DEVICE!

Thank you very much!

Best,
Di