commaai/openpilot #2739: camerad memory leak

ZwX1616, 2020-12-10:
**Describe the bug**

camerad uses 2-3% more memory per day if run for days

**How to reproduce or log data**

run camerad for extended amount of time

**Expected behavior**

constant memory usage

**Device/Version information (please complete the following information):**
 - Device: EON/comma two
 - <s>Dongle ID:</s>
 - <s>Route:</s>
 - <s>Timestamp:</s>
 - Version: current master
 - <s>Car make/model:</s>

**Additional context**

ZwX1616, 2020-12-11:
this might have been in camerad for ever. are you interested in looking into it? @deanlee

deanlee, 2020-12-11:
I would love to try

pd0wm, 2020-12-11:
Merged the PR. @ZwX1616 Can you check if the leak is gone?

ZwX1616, 2020-12-11:
> Merged the PR. @ZwX1616 Can you check if the leak is gone?

yeah the testing closet devices have been updated. we should be able to see the difference easily

adeebshihadeh, 2020-12-13:
Seeing no leak after two days, thanks @deanlee :)

closed 2020-12-13 by adeebshihadeh

deanlee, 2020-12-14:
I'm glad the problem was finally solved