commaai/openpilot #2740: Cruise Fault: Restart the car to engage

nojvek, 2020-12-10:
**Describe the bug**

**How to reproduce or log data**

Toyota Rav4 with EON running on black panda. There's roughly 1/3 chance that EON shows a "Cruise Fault: Restart the car to engage". Restarting the car doesn't fix it, restarting EON doesn't fix it. Once message appears you play game of dice to make it go away. Really annoying.

**Expected behavior**
CommaAIi software just works.

Don't see "Cruise fault: restart car to engage" error.

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]  EON 

 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
https://my.comma.ai/useradmin/?onebox=1bc766dcb82b0200

 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin] https://my.comma.ai/1bc766dcb82b0200/1607636895077/1607636971009

 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]: 1m into drive

 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)] release2

 - Car make/model: [e.g. Toyota Prius 2016] Toyota Rav4

**Additional context**

pd0wm, 2020-12-11:
Do you have the DSU unplugged?

Sometimes the fingerprinting takes to long to start emulating the DSU messages in time. The solution is to turn off the car and wait till completely turns off. The next drive it will be fine since the fingerprint is cached until the next boot of the EON. Rebooting the EON doesn't work since that makes it drop the cached fingerprint.

nojvek, 2020-12-16:
Yeah I had DSU removed. I mucked around for a while on the highway but wanted to reach my destination so I plugged DSU back in and proceeded my merry way.

pd0wm, 2020-12-16:
Closing this issue since this is kind of expect behavior for now. Hopefully we can make the firmware query somewhat faster in the future, but there are no short term plans for that.

closed 2020-12-16 by pd0wm

nojvek, 2020-12-16:
I understand you're closing this because of "priorities", but could we at-least make a task for this somewhere? 

comma's DSU-less feature doesn't work as expected.  This was not a great user experience being stuck on a highway and not being able to use any form of cruise control.

Kerman, 2021-04-04:
Deeply affected by this bug in 2017 prius prime. Also dsu disabled. It’s made worse by my unit not being able to hold a charge overnight anymore. Every morning is 3-10 reboots of my car to enable openpilot. :( thank you for the tip that re enabling dsu will get me Un stuck from the side of the road with no cruise!

davisdane2, 2023-04-20:
Same problem. 2017 Prius Prime with TSS-P. Its not a fun fix but what works for me is start the car with DSS plugged in and then I have my passenger unplug the DSS unit when im traveling around 5 mph. Then I get experimental controls.  Seems like the comma 3 has an established fingerprint and the fault will not appear. Must start the car with DSS plugged in or it will give cruise fault. Sometimes I need to reboot the comma with DSS unplugged to let the comma 3 allow experimental mode.

I'm on Openpilot master-ci 0.9.1 w/ comma 3