commaai/openpilot #2750: 2018 Kia Stinger 2.0L Premium "Can Error: Check connections" #installation-help told me to post here

oricalworks, 2020-12-12:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

adeebshihadeh, 2020-12-12:
Can you edit your post and fill out the bug report template?

oricalworks, 2020-12-12:
Describe the bug
Openpilot gives me "Can Error: Check connections" when everything is connected properly in my 2018 Kia Stinger 2.0L Premium.

How to reproduce or log data
Not really sure I've checked all connections and virtually Chris verified my connections were correct too. I just got my comma 2 today and installed it.

Expected behavior
Openpilot Engagement 

Device/Version information (please complete the following information):

Device: comma two
Dongle ID: dbad5babdf6cf346
Route: dbad5babdf6cf346|2020-12-11--17-46-20 , dbad5babdf6cf346|2020-12-11--13-13-52
Version: 0.8.
Car make/model: 2018 Kia Stinger 2.0L Premium

oricalworks, 2020-12-12:
> Can you edit your post and fill out the bug report template?

done :)

oricalworks, 2020-12-12:
Looking to buy whom ever fixes it a couple of coffees. :)

oricalworks, 2020-12-13:
Here is my fingerprint from workbench 67: 8, 127: 8, 304: 8, 320: 8, 339: 8, 356: 4, 358: 6, 544: 8, 576: 8, 593: 8, 608: 8, 688: 5, 809: 8, 854: 7, 870: 7, 871: 8, 872: 8, 897: 8, 902: 8, 909: 8, 916: 8, 1040: 8, 1042: 8, 1056: 8, 1057: 8, 1078: 4, 1107: 5, 1136: 8, 1151: 6, 1168: 7, 1170: 8, 1173: 8, 1184: 8, 1265: 4, 1280: 1, 1287: 4, 1290: 8, 1292: 8, 1294: 8, 1312: 8, 1322: 8, 1345: 8, 1348: 8, 1363: 8, 1369: 8, 1379: 8, 1384: 8, 1407: 8, 1419: 8, 1427: 6, 1456: 4, 1470: 8, 2000: 8, 2004: 8, 2008: 8, 2012: 8

robbaker92, 2021-01-01:
If you are still having this problem try to remove the car for line 333 of values.py.

oricalworks, 2021-01-03:
> If you are still having this problem try to remove the car for line 333 of values.py.
which car?
DBC = {
  CAR.ELANTRA: dbc_dict('hyundai_kia_generic', None),
  CAR.ELANTRA_GT_I30: dbc_dict('hyundai_kia_generic', None),
  CAR.GENESIS_G70: dbc_dict('hyundai_kia_generic', None),
  CAR.GENESIS_G80: dbc_dict('hyundai_kia_generic', None),
  CAR.GENESIS_G90: dbc_dict('hyundai_kia_generic', None),
  CAR.HYUNDAI_GENESIS: dbc_dict('hyundai_kia_generic', None),
  CAR.IONIQ_EV_2020: dbc_dict('hyundai_kia_generic', None),
  CAR.IONIQ_EV_LTD: dbc_dict('hyundai_kia_generic', None),
  CAR.IONIQ: dbc_dict('hyundai_kia_generic', None),
  CAR.KIA_FORTE: dbc_dict('hyundai_kia_generic', None),
  CAR.KIA_NIRO_EV: dbc_dict('hyundai_kia_generic', None),
  CAR.KIA_OPTIMA: dbc_dict('hyundai_kia_generic', None),
  CAR.KIA_OPTIMA_H: dbc_dict('hyundai_kia_generic', None),
  CAR.KIA_SORENTO: dbc_dict('hyundai_kia_generic', None),
  CAR.KIA_STINGER: dbc_dict('hyundai_kia_generic', None),
  CAR.KONA: dbc_dict('hyundai_kia_generic', None),
  CAR.KONA_EV: dbc_dict('hyundai_kia_generic', None),
  CAR.SANTA_FE: dbc_dict('hyundai_kia_generic', None),
  CAR.SONATA: dbc_dict('hyundai_kia_generic', None),
  CAR.SONATA_2019: dbc_dict('hyundai_kia_generic', None),
  CAR.PALISADE: dbc_dict('hyundai_kia_generic', None),
  CAR.VELOSTER: dbc_dict('hyundai_kia_generic', None),
}

robbaker92, 2021-01-03:
# these cars use the FCA11 message for the AEB and FCW signals, all others use SCC12
  "use_fca": set([CAR.SONATA, CAR.ELANTRA, CAR.ELANTRA_GT_I30, CAR.KIA_STINGER, CAR.IONIQ, CAR.IONIQ_EV_2020, CAR.KONA_EV, CAR.KIA_FORTE, CAR.KIA_NIRO_EV, CAR.PALISADE, CAR.GENESIS_G70, CAR.KONA]), look for this in values.py in Hyundai and remove the CAR.KIA_STINGER,

oricalworks, 2021-01-03:
Opps i was looking in release2 were you referring to master?

robbaker92, 2021-01-03:
Release2 is okay it’s a bit higher up the page than what you have got in your other comment

oricalworks, 2021-01-03:
@robbaker92 I added you to my fork, sorry for my stupidity if I'm mistaken.

oricalworks, 2021-01-03:
Done

oricalworks, 2021-01-03:
@robbaker92 Thanks for all your help I truly appreciate it!

robbaker92, 2021-01-03:
Let me know if you have any more problems.

oricalworks, 2021-01-03:
I'm still getting Can Error check connections

robbaker92, 2021-01-03:
You can try xxx979xxx’s 8.0 branch. Use “ cd /data/openpilot
git remote add xx979xx https://github.com/xx979xx/openpilot.git
git fetch --all” and then “git checkout 8.0”. After your need to and the fingerprint back in. Check if it’s made a difference, if not then try to remove the same thing from value.py.

oricalworks, 2021-01-03:
hmm I'm sure i didn't do any typos when adding a new array. On the new branch it endlessly hangs at this point
```
root@localhost:/data/openpilot$ ./launch.sh
Using interface 'wlan0'
OK
/data/openpilot/launch_chffrplus.sh: line 29: echo: write error: No such device
/data/openpilot/launch_chffrplus.sh: line 31: echo: write error: No such device
/data/openpilot/launch_chffrplus.sh: line 34: echo: write error: No such device
/data/openpilot/launch_chffrplus.sh: line 35: echo: write error: No such device
/data/openpilot/launch_chffrplus.sh: line 38: echo: write error: No such device
dinfo 1920x1080
egl version 1.4
egl w 1920 h 1080
gl version OpenGL ES 3.2 V@145.0 (GIT@I0ac3940325)
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
clang -o selfdrive/controls/lib/lateral_mpc/lib_mpc_export/acado_solver.os -c -std=gnu11 -DQCOM -mcpu=cortex-a57 -g -fPIC -O2 -Wunused -Werror -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -DQCOM -mcpu=cortex-a57 -fPIC -Iphonelibs/acado/include -Iphonelibs/acado/include/acado -Iphonelibs/qpoases/INCLUDE -Iphonelibs/qpoases/INCLUDE/EXTRAS -Iphonelibs/qpoases/SRC -Iphonelibs/qpoases -Iselfdrive/controls/lib/lateral_mpc/lib_mpc_export selfdrive/controls/lib/lateral_mpc/lib_mpc_export/acado_solver.c
clang -o selfdrive/controls/lib/longitudinal_mpc/lib_mpc_export/acado_solver.os -c -std=gnu11 -DQCOM -mcpu=cortex-a57 -g -fPIC -O2 -Wunused -Werror -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -DQCOM -mcpu=cortex-a57 -fPIC -Iphonelibs/acado/include -Iphonelibs/acado/include/acado -Iphonelibs/qpoases/INCLUDE -Iphonelibs/qpoases/INCLUDE/EXTRAS -Iphonelibs/qpoases/SRC -Iphonelibs/qpoases -Iselfdrive/controls/lib/longitudinal_mpc/lib_mpc_export selfdrive/controls/lib/longitudinal_mpc/lib_mpc_export/acado_solver.c
clang -o selfdrive/controls/lib/longitudinal_mpc_model/lib_mpc_export/acado_solver.os -c -std=gnu11 -DQCOM -mcpu=cortex-a57 -g -fPIC -O2 -Wunused -Werror -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -DQCOM -mcpu=cortex-a57 -fPIC -Iphonelibs/acado/include -Iphonelibs/acado/include/acado -Iphonelibs/qpoases/INCLUDE -Iphonelibs/qpoases/INCLUDE/EXTRAS -Iphonelibs/qpoases/SRC -Iphonelibs/qpoases -Iselfdrive/controls/lib/longitudinal_mpc_model/lib_mpc_export selfdrive/controls/lib/longitudinal_mpc_model/lib_mpc_export/acado_solver.c

```

robbaker92, 2021-01-03:
If your using workbench then try the install button in that use xxx979xxx fork and 8.0 branch

oricalworks, 2021-01-03:
Good Morning! I just reinstalled using workbench. It is still hanging on that process from last night.

robbaker92, 2021-01-03:
Reboot a couple of times

oricalworks, 2021-01-03:
Ok so I added my fingerprint and got CAN Error: check connections and same thing when I took CAR.stinger out of that "use_fca": set([CAR.SONATA, CAR.ELANTRA, CAR.ELANTRA_GT_I30, CAR.KIA_STINGER, CAR.IONIQ, CAR.IONIQ_EV_2020, CAR.KONA_EV, CAR.KIA_FORTE, CAR.KIA_NIRO_EV, CAR.PALISADE, CAR.GENESIS_G70, CAR.KONA]) and still got same Can error

robbaker92, 2021-01-03:
run the command "git submodule update --init" with two minus symbols before init. and then reboot.

oricalworks, 2021-01-03:
Ok done same issue still i'm going to upload a video of my setup.

oricalworks, 2021-01-03:
https://drive.google.com/file/d/17iSVhIJ8ldI6jruoDp6GL4f18aht5HOe/view?usp=sharing

robbaker92, 2021-01-03:
in useradmin is it showing a platform?

oricalworks, 2021-01-03:
it says kia stinger GT2
https://my.comma.ai/useradmin/?onebox=98b4e32baa5d21ec

oricalworks, 2021-01-03:
here is my dongle id 
98b4e32baa5d21ec

robbaker92, 2021-01-03:
can you make a drive public

oricalworks, 2021-01-03:
https://my.comma.ai/98b4e32baa5d21ec/1609597910177/1609599138412
https://my.comma.ai/cabana/?route=98b4e32baa5d21ec%7C2021-01-02--09-31-50&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F98b4e32baa5d21ec%2F3f9a50846fd0f8dfb69401d62541eeec_2021-01-02--09-31-50&seekTime=540

oricalworks, 2021-01-03:
did that work?

oricalworks, 2021-01-03:
im so stupid 
98b4e32baa5d21ec|2021-01-02--11-31-44 is now public

oricalworks, 2021-01-03:
here is another one 98b4e32baa5d21ec|2021-01-01--11-58-38

robbaker92, 2021-01-03:
Can't see anything that I think is making this error, sorry I can't be more helpful. Discord might have a fix.

oricalworks, 2021-01-03:
@robbaker92 Thanks so much for all your help and dedication! I really appreciate it!

oricalworks, 2021-01-03:
what channel should I post on?

robbaker92, 2021-01-03:
Give installation-help a go.

oricalworks, 2021-01-03:
once again thanks for your help

oricalworks, 2021-01-06:
@robbaker92 Hey I've made some progress and have a different error now but no can errors.
It says "openpilot unavailable: waiting for controls to start".
I'm using this repository which is a fork of the HKG community branch
https://github.com/Circuit-Pro/openpilot

oricalworks, 2021-01-06:
and I had an issue where my fingerprint was not correct. Here is the correct one 
67: 8, 127: 8, 304: 8, 320: 8, 339: 8, 356: 4, 358: 6, 544: 8, 576: 8, 593: 8, 608: 8, 688: 5, 809: 8, 854: 7, 870: 7, 871: 8, 872: 8, 897: 8, 902: 8, 909: 8, 916: 8, 1040: 8, 1042: 8, 1056: 8, 1057: 8, 1078: 4, 1107: 5, 1136: 8, 1151: 6, 1168: 7, 1170: 8, 1173: 8, 1184: 8, 1265: 4, 1280: 1, 1287: 4, 1290: 8, 1292: 8, 1294: 8, 1312: 8, 1322: 8, 1345: 8, 1348: 8, 1363: 8, 1369: 8, 1379: 8, 1384: 8, 1407: 8, 1419: 8, 1427: 6, 1456: 4, 1470: 8

oricalworks, 2021-01-06:
@xps-genesis @xx979xx  Hey would you guys possibly have any ideas? Any are greatly appreciated! :)

xps-genesis, 2021-01-06:
Have you tried my fork?

xps-genesis, 2021-01-06:
URL: https://github.com/xps-genesis/openpilot.git
Branch: xps

oricalworks, 2021-01-06:
Ill try it now

oricalworks, 2021-01-06:
@xps-genesis Ok so it says dashcam mode ill fork yours and add my fingerprint

xps-genesis, 2021-01-06:
Dashcam mode is not because of fingerprint. My fork would say fingerprint unrecognized if that was the issue. Did you turn on the community features toggle in settings?

oricalworks, 2021-01-06:
hmm let me double check

oricalworks, 2021-01-06:
yes community features are on. Still having same issue. "Dashcam Mode for unsupported car"

xps-genesis, 2021-01-06:
dm me on discord xps#8163

oricalworks, 2021-01-12:
@xps-genesis Hey just got the replacement. It appears to be working because car now has a slew of errors when its plugged it that it never had.

oricalworks, 2021-01-12:
https://drive.google.com/file/d/1BNwZ144cGKP9lRlDXnF90QWl_T4IvkHO/view?usp=sharing

oricalworks, 2021-01-14:
Fixed issue bad Hyundai type C harness with bad connection.

closed 2021-01-14 by oricalworks

JoshuaHarrisJBh, 2021-07-03:
> and I had an issue where my fingerprint was not correct. Here is the correct one
>
> 67: 8, 127: 8, 304: 8, 320: 8, 339: 8, 356: 4, 358: 6, 544: 8, 576: 8, 593: 8, 608: 8, 688: 5, 809: 8, 854: 7, 870: [...]