commaai/openpilot #2767: Manager Failed to Start

ziacivil01, 2020-12-12:
**Describe the bug**

**How to reproduce

ce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

pd0wm, 2020-12-14:
Is it connected to the internet? Registration could fail because the system time is wrong or because there is no internet connection at all. Try putting in a sim card just to get past the registration. It should only be needed once.

If that doesn't work, can you try a doing a factory reset?

nhi5758, 2020-12-14:
D

ziacivil01, 2020-12-14:
Thanks, I can do a factory reset but I do not see the button that can be pressed to reset it.

ziacivil01, 2020-12-14:
Thanks for the help. I was able to factory reset the service and it seems Okay. Here is the link to the video https://www.youtube.com/watch?v=0MPv_hSH3hk&ab_channel=ErichMoraga

closed 2020-12-14 by ziacivil01

selfdrivingmenee, 2021-09-08:
Identical issue with a fork, connected to the internet and pings correctly from shell. Tried to reset but no luck!