commaai/openpilot #2769: Openpilot calibration stuck at 0%

nlovlyn, 2020-12-12:
**Describe the bug**

I purchased the comma 2 for my 2018 rav4 xle about 1 month ago. I am having issues getting it to calibrate. What is happening is that I will get the screen that says 'calibration 0% drive above 15mph' and then nothing happens. No errors and no changes in percent even after driving for 1hr. I have tried versions 0.7.9, 0.7.10, and 0.8.0 with community features both enabled and disabled for each version and I get the same results. I have tried unplugging everything and re-plugging. I've looked through the logs and it looks like almost everything under 'liveCalibration' is zero and does not update. The community says that the mounting looks good from the photos and dashcam recording I posted to discord. 

**How to reproduce or log data**

1. Start the car
2. Wait for calibration screen to display
3. Drive for a while but no change in percent or errors

Log Data: all 'liveCalibration' blocks appear to have the same values seen below.

> ( logMonoTime = 11086766343778,
  liveCalibration = (
    calStatus = 0,
    calCycle = 0,
    calPerc = 0,
    extrinsicMatrix = [0, -1, 0, 0, 0, 0, -1, 1.22, 1, 0, 0, 0],
    rpyCalib = [0, 0, 0],
    rpyCalibSpread = [0, 0, 0],
    validBlocks = 0 ),
  valid = true )

**Expected behavior**

I expect it to calibrate or display an error that it can not calibrate. 

**Device/Version information (please complete the following information):**
 - Device: Comma 2
 - Dongle ID: 0feb5791ebfe2272
 - Route:  0feb5791ebfe2272|2020-12-09--17-11-23 [0.8.0], 	0feb5791ebfe2272|2020-11-06--15-40-51[0.7.9]
 - Timestamp: Entire drive. Jump to the middle to
 - Version: I have tried versions 0.7.9, 0.7.10, and 0.8.0 with community features both enabled and disabled for each version
 - Car make/model: 2018 Toyota RAV4

**Additional context**

nlovlyn, 2020-12-13:
Clip from one of my drives to show mounting:
https://discord.com/channels/469524606043160576/524592892627517450/774785986617671712

Discord troubleshooting sessions:
1.  https://discord.com/channels/469524606043160576/524592892627517450/774453306642530304
2.  https://discord.com/channels/469524606043160576/524592892627517450/775513783350984704

adeebshihadeh, 2020-12-13:
Nice work looking through the logs. Calibration is always 0 because modeld crashed. If you've tried multiple versions of openpilot, I'd suspect this is a HW issue. Contact support@comma.ai and we'll take care of it for you.

nlovlyn, 2020-12-13:
Thank you, I tried to dive as deep as I could with my knowledge before opening on an issue that I could resolve! 😄  I'll reply to support and ask for a new kit! When I originally reached out to them they pointed me here. I will update with what they end up saying.

In the meantime is there anything I can do to add additional logging to the calibration python script to help pinpoint the root cause of this incident or would that even be worth digging into?

Cheers,
Nick

pd0wm, 2020-12-14:
Would be interesting to see if there are any tombstones in `/data/tombstone` related to the modeld crashes. But we can also take a look when we get your device back.

adeebshihadeh, 2020-12-17:
Closing as this seems to be a HW issue, but if you'd really like to debug, look into modeld, not calibrationd.

closed 2020-12-17 by adeebshihadeh

nlovlyn, 2020-12-22:
I received the new device and it calibrated right away and I was able to engage openpilot! Thanks for everyone's help with this!