commaai/openpilot #19497: A

Odin61, 2020-12-14:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

mpptrainer, 2020-12-14:
i would like to cancel my registration and possibly be refunded

                Dott- Mauro Panzeri
   Laureato in Scienze Motorie & Sport
                  +39 320 555 94 23

closed 2020-12-14 by adeebshihadeh

pd0wm, 2020-12-14:
@mpptrainer Please contact support@comma.ai for refunds