commaai/openpilot #19513: Spontaneous disengagement/reboot

schultp, 2020-12-16:
**Describe the bug**

Cruising on expressway and OP spontaneously disengaged and the comma2 rebooted

**How to reproduce or log data**

Single occurrence. No steps at this time to reproduce. 

**Expected behavior**

No spontaneous disengagement of OP

**Device/Version information (please complete the following information):**
 - Device: comma2
 - Dongle ID: 0a5c7571d1d1b4a8
 - Route: 0a5c7571d1d1b4a8|2020-12-14--17-15-38--14
 - Timestamp: At 17:29 - 17:30
 - Version: 0.8.0
 - Car make/model: 2017 Chevrolet Volt

**Additional context**
Running OBD harness.

pd0wm, 2020-12-16:
Thanks for the report. We'll investigate the logs!

My current theory is the cellular modem driver sometimes locks up and triggers the watchdog. This is something we're investigating, and doesn't have an easy fix. This should be a very rare event though, so if you see it again let us know.

Jon8RFC, 2021-01-10:
I have some random reboots to share, but I don't recall when they were.  I'm happy to manually look through all of my logs and narrow it down if you can tell me which search term(s) to look for.

Device: comma2
Dongle ID: 03690f0d837307af
Route: 
Timestamp: 
Version: 0.8.1
Car make/model: 2018 Hyundai Sonata Limited Ultimate

Jon8RFC, 2021-01-17:
@pd0wm

**Describe the bug**

Rebooted idiopathically.

**How to reproduce or log data**

No known way to reproduce.

bootlog after the reboot:
03690f0d837307af|2021-01-16--15-22-35--0

**Expected behavior**

No random reboot.

**Device/Version information (please complete the following information):**
 - Device: comma2
 - Dongle ID: 03690f0d837307af
 - Route: 03690f0d837307af|2021-01-16--15-09-09--12
 - Timestamp: at the end of this route segment
 - Version: 0.8.1
 - Car make/model: 2018 Hyundai Sonata Limited Ultimate

**Additional context**

My car beeped twice quickly, no more than 10 seconds before the reboot.  It sounded like the seatbelt warning beep.
The cruise button was probably toggled on, but OP was not engaged.

Jon8RFC, 2021-01-19:
Random reboot as it's sitting here on the desk, plugged into a reliable 5v power supply (not the computer).
My uploads aren't going for some reason, but the random reboot was at 2021-01-19 at 00:15am.

adeebshihadeh, 2021-01-19:
Yes, this is possibly a hardware issue. We'd be happy to swap it our you, and if it doesn't turn out to be your hardware, we can look into it more. Can you contact support@comma.ai?

pd0wm, 2021-02-01:
Closing this issue since it probably was hardware failure. Keeping https://github.com/commaai/openpilot/issues/19874 open for software issues.

newjacky2419, 2021-02-07:
**Describe the bug**
 freeze and rebooted randomly.

**How to reproduce or log data**

No known way to reproduce.

**Expected behavior**

No random reboot.

**Device/Version information (please complete the following information):**

Device: comma2
Dongle ID: fe30a344d9e657bd
Route: （I don’t know how can I found it)
Timestamp: （I don’t know how can I found it)
Version: 0.8.1 (Dragonpilot)
Car make/model: 2020 Toyota Prius Prime

**Additional context**
When my comma two freeze I took cellphone to record. 

https://youtu.be/yotG_RFBYwQ

As you can see this time comma two reboot many times(4 times). 10 minutes later. It freeze and reboot again.

https://youtu.be/uNi_f-DNSDY

This time just reboot one time.

Is my Comma Two hardware failure? Or it just the bug?

pd0wm, 2021-02-08:
If it happens a lot it's usually a hardware failure. But try running openpilot release instead of dragonpilot before drawing any conclusions.

amano3001, 2021-02-17:
Hi, I just had this happen to me on a fresh install of a brand new comma 2 device. 2017 Prius. It happened between 4pm and 5pm today.

newjacky2419, 2021-02-17:
> If it happens a lot it's usually a hardware failure. But try running openpilot release instead of dragonpilot before drawing any conclusions.

I reinstalled Openpilot. But this time is worse than before. Almost like #19874 issue. Today I drove on the freeway. My comma two freeze and rebooted while engaged. When comma two restart completely, I still saw my LTA failure warning on my dashboard.

VirtuallyChris, 2021-02-17:
@amano3001 and @newjacky2419 please email support@comma.ai so we can get these devices back and get working units to you.

michaelhonan, 2021-02-18:
Had this Yesterday. Route ID (crash should be at the very end, or start of next segment): 4d70bc5e608678be|2021-02-17--14-14-39--3

It was a 35C/95F day, and my comma two froze then rebooted. Exactly as described here.

adeebshihadeh, 2021-04-14:
Closing this and tracking software related reboots in #19874.

closed 2021-04-14 by adeebshihadeh