commaai/openpilot #19516: v0.8.1 Japan model 2021 RAV4 Hybrid G TSS2 CAN error

bcredford, 2020-12-16:
On rav4-fw branch due to fingerprint conflict. Installation triggers multiple dashcam alarms and displays CAN error: check connections on Comma 2 UI. tmux ssh logs provided to support desk and avail upon request.

VirtuallyChris, 2020-12-16:
please properly fill out the bug report template

bcredford, 2020-12-16:
Didn't see template. What additional info does it need?

Brian Redford

VirtuallyChris, 2020-12-16:
You'll see it when you go to create a bug report. It has a template to fill out.

VirtuallyChris, 2020-12-16:
want to try master-ci again?

adeebshihadeh, 2020-12-17:
Verified with one of your routes that your car now correctly fingerprints. If you switch to master-ci, your device will recognize the car correctly.

bcredford, 2020-12-17:
Oops just replied to Chris's email. Master-ci not working for me. Getting a txmt buffer error on tmux a.

bcredford, 2020-12-17:
Still getting can error on master-ci and transmit buffer full error on tmux a. Got a video if you have somewhere to drop. 183mb.

Closed #19516<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fcommaai%2Fopenpilot%2Fissues%2F19516&data=04%7C01%7C%7Cee9cb477fcfe449890f308d8a2250cd3%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637437627300065106%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=IgNEkF2DQVIs3vhl6xYv2mJpA2Anh65sRBImNyG%2Frjo%3D&reserved=0>.

bcredford, 2020-12-17:
Why did you close this? The issue isn't resolved.

bcredford, 2020-12-17:
Thank you

Reopened #19516<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fcommaai%2Fopenpilot%2Fissues%2F19516&data=04%7C01%7C%7Ca8179465b72d49ef060608d8a22e8b0c%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637437668070259636%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=dqMuq%2B%2Bsp0m3T1%2BLbZjLdc%2F801Cnmgkb6T4i4jhVTks%3D&reserved=0>.

VirtuallyChris, 2020-12-17:
had a bad USB-C cable. good to close.

closed 2020-12-17 by VirtuallyChris