commaai/openpilot #19524: Device Tab Causes "Unfortunately BaseUI has stopped"

C0mpl3t3N00b, 2020-12-16:
**Describe the bug**

On Device, tapping on settings icon and tapping on Device tab will cause "Unfortunately BaseUI has stopped"

**How to reproduce or log data**

Tap Settings icon, tap on Device tab

**Expected behavior**

Tapping on Device tab should allow me to use functions in said tab

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: 
 - Route: n/a
 - Timestamp: n/a
 - Version: https://github.com/commaai/openpilot/commit/5138d213a75a01f9ba1855131aa4cd712a3a3ce6
 - Car make/model: n/a
**Additional context**

adeebshihadeh, 2020-12-16:
What's your Dongle ID? Found in Settings -> Device.

C0mpl3t3N00b, 2020-12-16:
Well if you read above I cannot access the Device menu, but luckily I found it from a previous issue I submitted: bdad50b8a1f9fa8e

adeebshihadeh, 2020-12-16:
We fixed a bug that was on master recently that caused this issue, but you seem to have the fix. Can you reproduce the issue and paste the output of `logcat -d` in something like a github gist?

C0mpl3t3N00b, 2020-12-17:
> We fixed a bug that was on master recently that caused this issue, but you seem to have the fix. Can you reproduce the issue and paste the output of `logcat -d` in something like a github gist?

I ended up fully uninstalling OP, reinstalling release2, and now back on master-ci. Can no longer reproduce and will close issue. Please reopen if you feel it needs further investigation.

closed 2020-12-17 by C0mpl3t3N00b