commaai/openpilot #19531: Zygote restarting while OP active

michaelhonan, 2020-12-17:
**Describe the bug**

**How to reproduce or log data**

So for the past couple months, after a couple days of uptime, the comma two offroad UI will glitch out. The buttons respond with highlighting upon touch, but everything else stops working.

Starting the car correctly starts openpilot, and all works just fine, except the sound stops working completely (no disengage/engage sound).

A reboot solves the issue, until it eventually occurs again a few days later. Factory resets/reflashing have not fixed the issue, and I don't believe it to be a hardware fault.

---

This time, I left the comma two to bask in its glitched state and this ended up happening; the comma two had the spinning logo, while ALSO still driving my car. In the video below, I nudge the wheel to cause on purpose ping pong to prove it was still steering.

**Expected behavior**

The above behaviour should not occur. The UI should remain intact, and the device shouldn't "restart" while openpilot is active.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 4d70bc5e608678be
 - Route: 4d70bc5e608678be|2020-12-13--12-52-39--5 (When I can reproduce again, I will get logs... this route isn't helpful (adeeb looked at it).
 - Timestamp: N/A
 - Version: `master` based fork (basic additions like tuning), was rebased onto `1ea307a2996f16104501bf1d497e202326000dc1`
 - Car make/model: 2018 Subaru Impreza

**Additional context**

Videos of unresponsive UI / restart while driving.

[Unresponsive UI](https://cdn.discordapp.com/attachments/532205411709747212/787559036626534430/IMG_0348.MP4)
[comma two restarting during drive (still engaged -  WARNING: Colourful language)](https://cdn.discordapp.com/attachments/532205411709747212/787559363770056744/IMG_0362.MP4)

cydia2020, 2021-02-27:
Seeing the same thing here, this seems to be different from #19874 as in this situation, the drive continues (openpilot still engages, steers, and sends acceleration commands) but the UI crashes and the boot logo shows.

adeebshihadeh, 2021-05-21:
I have a theory on what causes this, and I believe this should be better in 0.8.3+. Have you guys since this since then?

cydia2020, 2021-05-21:
Can confirm that I haven't encountered this for a while now.

adeebshihadeh, 2021-05-21:
@cydia2020 did you experience it somewhat consistently previously?

cydia2020, 2021-05-21:
> @cydia2020 did you experience it somewhat consistently previously?

No, only experienced it twice.

michaelhonan, 2021-05-21:
I consistently experienced it, and haven't for quite a while.

adeebshihadeh, 2021-05-24:
I believe these crashes were due to SystemServer's processes getting killed to free up memory. This likely affected users who left their devices in their cars and drove just often enough to not hit the thresholds in thermald to shut off the device. A significant leak was fixed in 0.8.3, so once we have better logging around this (#20978), I'll close the issue.

adeebshihadeh, 2021-07-12:
I think this class of zygote crashes is fixed. Tracking a new class of zygote crashes in #21539.

closed 2021-07-12 by adeebshihadeh