commaai/openpilot #19543: Car unrecognized Toyota Highlander 2019

grevg, 2020-12-19:
**Describe the bug**

After installing comma two and connecting the display says Dashcam mode Car unrecognized 

**How to reproduce or log data**

**Expected behavior**

This model is in the list of supported, so I ordered from online shop comma.ai

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 922bcfdc7c1e117d
 - Route: 922bcfdc7c1e117d |2020-12-19
 - Timestamp: [19 December 
 - Version: 0.8
 - Car make/model: Toyota Highlander 2019 safety sense  US manufacturing for  Russian market 

**Additional context**

adeebshihadeh, 2020-12-19:
Fingerprinted added in d6caca943308ff96c96f6750cb7763f577240925. If you switch to master or master-ci, your car should now be recognized.

closed 2020-12-19 by adeebshihadeh

grevg, 2020-12-20:
> Fingerprinted added in [d6caca9](https://github.com/commaai/openpilot/commit/d6caca943308ff96c96f6750cb7763f577240925). If you switch to master or master-ci, your car should now be recognized.

Thanks for fast fix.
How do I switch to master? I have v 0.8.0 automatically installed from setup process.