commaai/openpilot #19545: 2021 toyota highlander hybrid unrecognized

kayakkid, 2020-12-19:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

kayakkid, 2020-12-19:
device 8ed6eb92fecf04de
car 2021 toyta highlander hybrid
new install 12-19-2020
version 8.0

vanillagorillaa, 2020-12-20:
It looks like your firmware versions for your vehicle will be included in the 0.8.1 release (currently in the devel branch)

closed 2020-12-20 by vanillagorillaa