commaai/openpilot #19579: Video frame dropping

dingyf12593, 2020-12-23:
Device: [EON Gold]
Dongle ID: [3f84146622df5f39]
Version: [0.8.1]
Car make/model: [ golf 7.5 2018]

Video frame dropping ，reinstall  twice，problem alwaysl

https://user-images.githubusercontent.com/6504134/102968705-55313600-452f-11eb-8183-00ac2b81f9d7.mov

nickgeurnop, 2020-12-23:
**Issue:** 
Comma Two display is dropping frames or appears "laggy." However, ALC and ACC operation is unaffected. Version 0.8.1 performs exceptional.

**Expected Behavior:** 
Comma Two display should relay real time driving footage uninterrupted and smoothly.   

Device: Comma Two
Dongle ID: 1615388ccbe3edc9
Version:  version 0.8.1 release2
Car make/model: 2019 Toyota Corolla Hatchback XSE (TSS2.0)

**Addition Notes**
Behavior occured as soon as I updated from 0.8.0 release2 --> 0.8.1 release2. I tried uninstalling and reinstalling once with behavior unchanged. 

https://user-images.githubusercontent.com/74438392/103042622-37bc9600-4548-11eb-905e-f328d936f84f.mov

davetapley, 2020-12-23:
Same as ⬆️ , can post more info if needed?

adeebshihadeh, 2020-12-24:
How consistently is this happening? @dingyf12593 your routes aren't fully uploaded so I couldn't look into it. However, I looked at a segment from one of @nickgeurnop's latest routes which was normal. I suspect this is isolated to UI, have any of you felt any degradation in driving performance when it happens? I can look into this more with specific routes and segments that are fully uploaded.

dingyf12593, 2020-12-25:
After upgrading to 0.8.1, the total driving distance is expected to be 40 km. This situation has been happening all the time, and the driving performance has not decreased @adeebshihadeh

dingyf12593, 2020-12-25:
le pro3 6+64GB ，Is it possible that the memory version is different ，If most people are normal

nickgeurnop, 2020-12-25:
@adeebshihadeh it occurs all the time, when openpilot is engaged or not. 

Driving performance is unaffected as far as I know. In fact, 0.8.1 is a great improvement over 0.8.0 in my experience. 

My only real concern would be video playback quality if I needed dashcam footage. Other than that it just looks weird since you can't turn off the screen. Hope it is an easy fix! Thank you for looking into this!

adeebshihadeh, 2020-12-25:
This doesn't affect the recorded video. Can one of you test #19589 and let me know if it fixes the issue?

dingyf12593, 2020-12-25:
@adeebshihadeh  ok, try in tonight , now, im at work

dingyf12593, 2020-12-25:
Fixed @adeebshihadeh

dingyf12593, 2020-12-25:
No takeover status

https://user-images.githubusercontent.com/6504134/103143402-15956480-4750-11eb-9575-011e0854c0a2.mp4

adeebshihadeh, 2020-12-28:
@dingyf12593 can you send me the specific segments and routes those videos are from?

loveloveses, 2020-12-31:
> Device: [EON Gold]
> Dongle ID: [3f84146622df5f39]
> Version: [0.8.1]
> Car make/model: [ golf 7.5 2018]
> 
> Video frame dropping ，reinstall twice，problem alwaysl
> 
>  video0.mov

是国内的豆友么？设备是官方的么？
我今天用酷派的设备刷机到081以后也出现这个状况，所以，你的设备是国内的还是官方的？

deanlee, 2020-12-31:
@adeebshihadeh Take a look at https://github.com/commaai/openpilot/pull/19629, it may be a poll problem

adeebshihadeh, 2021-01-12:
We haven't been able to reproduce this issue, so closing due to inactivity.

closed 2021-01-12 by adeebshihadeh