commaai/openpilot #19582: Haotic gas/brake comands Toyota Highlander 2019 -master-ci

grevg, 2020-12-23:
**Describe the bug**

When stopped in trafic the OP seems to have chaotic distance measurements from the stopped car in front, these leads to chaotic brake system engagements and disengaments which can be seen from specific noise of brake system, car micro movements, headlights darkening (from the power consumption of brake fluid pump)

**How to reproduce or log data**

Stop in traffic with OP engaged

**Expected behavior**

Some "smoothing" of distance data or gas/brake commands needed in the near zero speed modes.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 922bcfdc7c1e117d
 - Route: [e.g. 922bcfdc7c1e117d|2020-12-21--11h-13m
 - Timestamp: 922bcfdc7c1e117d|2020-12-21--11h-13m
 - Version: master-ci 12/21/2020
 - Car make/model: Toyota Highlander 2019 safety sense, derated to 249 hp

**Additional context**

Also at speed around 60 kmph with 8 mode automatic transmission same reason of unsmoothed gas commands lead to chaotic change of switching between 6,7,8 transmission mode (since the car cant understand if you are moving steadily in eco mode or trying to accelerate). Also some smoothing required.

VirtuallyChris, 2020-12-24:
Is your DSU unplugged?

grevg, 2020-12-24:
> Is your DSU unplugged?

DSU is in place, I haven’t unplugged anything 
And I was actually surprised OP continues to work under 40km/h and even stops 
First day I tried OP turned off below 40 km/h as the stock ACC did before , now it doesn’t turn off at low speeds

VirtuallyChris, 2020-12-24:
If your DSU is plugged in then you are still using stock ACC - meaning the behavior you encountered is due to the Toyota system. openpilot will only control your car ACC if you unplugged your DSU but this isn't an encouraged configuration since you'd lose AEB.

closed 2020-12-24 by VirtuallyChris

grevg, 2020-12-24:
> If your DSU is plugged in then you are still using stock ACC - meaning the behavior you encountered is due to the Toyota system. openpilot will only control your car ACC if you unplugged your DSU but this isn't an encouraged configuration since you'd lose AEB.

My stock DSU has no ACC feature of working below 40 km/h (not stop and go in traffic feature as well), while with OP engaged i am able to drive down to 0 speeds, stay in traffic and than drive forward again following the car in fron (your website tells that this is not possible at all with DSU in place for this car model. This below 40 km/h engaments is not always possible, some times car not reacting to SET, sometimes it does. 
at speeds of 60-80 km/h the ACC is behaving completetly differently from the stock ACC behavior i had before installation of OP (chaotic gas increase decrease which leads to gear shifting with steady speed following the car in front).
I guess it is not toyota giving gas/brake commands in these situations or there is some interference between commands of DSU and OP.