commaai/openpilot #19613: Security hardening

capstarall, 2020-12-29:
**Local Hardening**
Sure, “it’s just a devkit” for now, understood. But it would be wise to get a headstart on hardening the system so that when it’s driving 10k+ cars, it’s got a reliable and tested security model.

While by default the remote / network attack surface is quite limited (kudos for that), here’s a few suggestions to make things better yet still functional, primarily for the local environment, in order to limit the blast radius and make practical attacks non-trivial for anyone who manages to get some sort of internet or wifi rw/x access on the device (via comma’s services or otherwise).

**/dev/shm permissions**
777 is a little too open. Would 700 or 770 still let the necessary services read and write without disruption (some testing was done here, but more needed), while maintaining some sense of defense-in-depth?

Eg. if some random service did get compromised on the device, poking at stuff in /dev/shm seems like one of the things we’d only want comma components / root to do.

**Processes should run as non-root**

So that if there are ever bugs in these processes, they don’t automatically lead to a root shell. Seems like adding/reusing a local sandboxed user and doing a little work to the local services to make sure everybody is still communicating ok would be the easiest approach.

**SSH key should not be public**
Picking the SSH key off of the openpilot github is great for easy mode, but it probably should be randomly generated either 1) per device or 2) per openpilot install so you’re less scared of connecting this thing to the wifis.

**Installation warning on URLs other than openpilot official**
Measures should be taken to warn users in case they typo or are otherwise led to use a URL that looks or sounds like the official openpilot one (we all know the comma two is meant to use openpilot.comma.ai, it can still be non-obvious just with an addition confirmation from the other if they want to really install something else).

One typo or bad URL shouldn’t be a straight shot to running a backdoored image.

Anyways, hope this is helpful! Would be great to get to know if some of these were already on the list of things to do, or if they would even be items that have a shot on the current roadmap.

Really love your product. Keep up the chill work!

abdulrahman004, 2020-12-29:
> Installation warning on URLs other than openpilot official

We already get a warning of Warning: Untested branch for any non-official branch.

pd0wm, 2020-12-29:
We mainly focus our efforts on preventing code execution on the device. If you get code execution all bets are probably off, unless you're going to spend absurd amount of time sandboxing everything. That time is better spent on the former.

My thoughts on your questions:

1) /dev/shm has to be read/write by both openpilot and offroad (the APK). Apps run with very low permissions on android, so therefore we make /dev/shm writable by everyone. When we switch offroad to qt instead of an APK in the near future we can revert the changes to the permissions of /dev/shm.
2) This should be doable. Some services still need some higher privileges which are annoying to get around. Off the top of my head: updated needs to mount an overlayfs, boardd needs to set the time and thermald/athenad needs to be able to reboot the device.
3) The default ssh key has an IP filter on it, and only works over wifi, not through athena/cellular already reducing the surface. Also SSH is turned off by default. If you turn on SSH you should really enter your github username, that disables the default key.
4) The C2 is a devkit, and should be able to run everything. Not really related to openpilot itself.

capstarall, 2020-12-29:
Thanks for your thoughtful replies!

Here's a few follow-ups that hopefully it won't take too much of your time.

@abdulrahman004

> We already get a warning of Warning: Untested branch for any non-official branch.

Sure, but this is more of an untested software warning and non-security related. As far as I can tell, anyone can provide a git URL and it will install the repo.

Maybe framing the scenario like this will help with the visualization.

What is protecting users against a campaign where anyone can register a similar github name, spreading a bunch of ads and articles proliferating spoofed installation URLs like comaai.com or github.com/comaai/openpilot, clicking the Fork button and making some small changes to the new repo and getting new users to install a backdoored openpilot?

@pd0wm

> 1. /dev/shm has to be read/write by both openpilot and offroad (the APK). Apps run with very low permissions on android, so therefore we make /dev/shm writable by everyone. When we switch offroad to [...]

ACK that apps do run with low permissions, but as you know RWX everyone means just that: if any app gets compromised, they can make changes to /dev/shm/*.

I didn't test it live, but may I ask to what affect could a rogue process have when writing to eg. `can`, `gpsLocation`, `liveCalibration` or others? Would the panda not allow any weird driving moments, or what exactly are it's limitations?

>     2. This should be doable. Some services still need some higher privileges which are annoying to get around. Off the top of my head: updated needs to mount an overlayfs, boardd needs to set the time and thermald/athenad needs to be able to reboot the device.

Understood. Could this be solved using Linux capabilities or native Android permissions? Not sure if Python complicates things significantly or not in this environment.

>     3. The default ssh key has an IP filter on it, and only works over wifi, not through athena/cellular already reducing the surface. Also SSH is turned off by default. If you turn on SSH you should really enter your github username, that disables the default key. At some point we might remove the default key, but for now it's very useful to debug issues with installer and device testing in production.

Yes, SSH is off by default and the IP filter is great and both of these do prevent an accidental "expose your car to the Internet". That being said, as soon as you connect to WiFi with SSH enabled, anyone on your local network can use the public key to drop into a root shell. Of course most consumers will use their home network, but would the guidance be to never use other networks or public WiFi either?

For some reason, I overlooked the documentation for how connecting your github username works with regard to the default key. Could you link me / briefly explain?

>     4. The C2 is a devkit, and should be able to run everything. Not really related to openpilot itself.

Sure, agree for the most part that the C2 should not be restricted. But see my top response to @abdulrahman004's comment on how easy it would be to spoof installation URLs. If a quick regex determines that they didn't type in the official repo URL, would a warning dialog box prompt before continuing make sense here? That seems like it would be pretty effective at stomping out spoofing type attacks.

> As a footnote:
> There is still the panda between the C2 and the car which will always maintain the safety model no matter what code the C2 is running. At some point we will lock the bootloader in the panda allowing only signed code to run (requiring a physical paw/key to unlock the bootloader).

Very interesting! Really appreciate the additional details.

pd0wm, 2021-04-08:
Quick update: 
 1) The ui is now completely replaced by Qt running under the same user as openpilot. This should allow us to get rid of all the permission hacks. When we have some time for this we'll go over the codebase and remove explicit permissions.
 2) Openpilot can now mostly run as a non-privileged user. In places where we need more permission we use `sudo` explicitly, so no need to run anything as root. We currently still need sudo to mount an overlayfs in updated, set cpu governors, reboot and set the system time based on panda RTC time.
  3) We removed the default SSH key in the latest NEOS update, and SSH is now turned off by default. We require the user to turn this on manually now, and can only be used with their public keys from their github account.
  4) Still considering a won't fix.

closed 2021-08-29 by pd0wm