commaai/openpilot #19623: BSM Lane Change Fault

cowanhm, 2020-12-30:
**Describe the bug**

2021 RAV4. The Lane Change BSM has a fault scenario. This vehicle has a “BLINKING” lane change marker that may allow BSM to NOT override the Lane Change Assist, depending on instantaneous status of signal when turn signal is turned on. 

**How to reproduce or log data**

Activate lane change while OP is engaged. If a vehicle is present or approaching, the “blocked” message comes up and prevents lane change. Once in awhile, the BSM Light on side-mirror appears to BLINK and vehicle starts lane change despite another vehicle being in blind spot. 

**Expected behavior**

The vehicles BSM status should prevent lane change, even is mirror indicator is blinking.

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: downloading
 - Route: downloading 
 - Timestamp: downloading
 - Version: 0.8.1 master 
 - Car make/model: 2021 RAV4 TSS 2.0

**Additional context**

cowanhm, 2020-12-30:
A minimum (0.5sec?) lockout delay upon blindspot-detect should solve issue.

cowanhm, 2020-12-31:
[image]

pd0wm, 2021-01-04:
Is there another bit that signifies the blinking state?

pd0wm, 2021-01-04:
Is it actually blinking or just losing track of the car for a little bit?

cowanhm, 2021-01-04:
After looking for the other bit, now I’m agreeing that it’s probably just losing and regaining. The physical light on the car appears to be a three state. Off,  Solid, or R Rapid Blinking. 

 It’s a little difficult to test... because it may scare people when I turn on the signal when they’re in my blind spot.

Thanks,
Cowan Moore
330-701-3922

cowanhm, 2021-01-04:
This discusses that third blinking state.  BSM light will blink if there’s a car in the Blindspot, and the turn signal is activated in that direction. I’ll try to re-simulate it this morning and send you the cabana

Thanks,
Cowan Moore
330-701-3922

pd0wm, 2021-01-04:
Thanks! My feeling it's just losing the car. Not sure if we want a cooldown on the signal, that could be annoying if you try to change lanes right after a car passes you.

ps11918, 2021-01-04:
Even 1 second delay could be helpful...   the BSM appears to fully deactivated prior to passing car being fully passed.

cowanhm, 2021-01-04:
Even 1 second delay could be helpful... the BSM appears to fully deactivate prior to passing car being fully passed.

adeebshihadeh, 2021-01-22:
I don't think we should be writing hacks like this to overcome deficiencies in the car's built in BSM. These delay values may not be consistent between models in a make.

closed 2021-01-22 by adeebshihadeh