commaai/openpilot #19631: [2018 Nissan Leaf] Occasionally disengages when lead car brakes

willemcvu, 2021-01-01:
**Describe the bug**

Occasionally, when following a lead car, using the Nissan ProPilot stock cruise control doing logntiunal control and OpenPilot doing steering, when the lead car brakes, OpenPilot/ProPilot completely disengage with the seatbelt unplugged message shown on the HUD.

**How to reproduce or log data**

1. Have openpilot activated for steering, with Nissan stock system doing longitudinal control.
2. Have lead car indicated on HUD with set speed faster than lead car speed.
2. Lead car brakes.

**Expected behavior**

Leaf should brake along with lead car and stay engaged, as it does 9/10 times.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: e4d79cf6b8b19a0d
 - Route: e4d79cf6b8b19a0d|2021-01-01--17-33-17
 - Timestamps: 17:52:07,  17:54:10, 17:55:44
 - Version: 0.8.1
 - Car make/model: 2018 Nissan Leaf SV w/ tech package (ProPilot)

**Additional context**
It hasn't been happening recently (it seems like either it happened more during the warmer months OR happened more on pre-0.8 OpenPilot), but it happened 3x on this one drive.

I submitted a bug report about this several months ago but never responded with a route, apologies for that. I'll close that and link to this.

vanillagorillaa, 2021-01-22:
It looks like the route that has been provided is no longer on the server. If you can get another route with this occurring please let if fully upload, then through useradmin (my.comma.ai/useradmin), click "preserve" on the route. 

Also, does this also occur without openpilot? You can set the toggle in settings for "enable openpilot" to off and record a drive.

willemcvu, 2021-03-02:
@vanillagorillaa I got another route with the issue occuring uploaded and preserved.

No, this does not happen without OpenPilot. I can tell OpenPilot initiates the disengagement by the seatbelt warning shown on the HUD.

willemcvu, 2021-03-02:
Looks like the brake pedal message isn't quite right, and is what is disengaging OpenPilot. I did NOT press the brake pedal here; this is the Nissan ACC engaging the friction brakes.

My guess is that most of the time, friction brakes are not really used much on the Leaf, thus, this issue is sporatic and happens more at high state of charge (when regen can't be used as heavily).

[image]

willemcvu, 2021-03-02:
OK, I think I've figured out what's going on with the help of the following route, as the first section is down a big mountain with lots of regen and occasionally friction brakes because it is so steep: e4d79cf6b8b19a0d|2021-02-28--11-40-33

Please note: I have a fork of OpenDBC that has the messages needed to make the graphs below)

Observations:

* 2b0, 4th byte, 2nd bit looks to be 1 when the brakelight is on and 0 when the brakelight is off. The brakelight on the Leaf engages when the brake pedal is pressed, or, with e-pedal engaged, when the accelerator pedal is fully released. The downhill section of the above route was driven entirely with e-pedal off in B mode, so anytime the brake light is engaged, the driver pressed the brake pedal.

* The other relevant braking-related messages look to be 1cb and 1cc.
* Approx 440-450 in the above log has very interesting data. It looks to me like the car was going down the mountain, and then hit a small bump which disabled the regen (the car disables regen anytime any grip is lost to avoid braking on 2/4 wheels).

[image]
[image]
[image]

Assuming I have correctly identified the above messages, it looks the brake pedal signal we were using is really the braking system hydraulic pressure, and so when the braking system mixes in friction braking (which happens more at high SOC and when grip is lost momentarily, among other factors), the "brake pedal" signal was tripping of OpenPilot.

I think we should be able to just increase the brake pedal disengage minimum value to above what the brake controller will ever mix in and we should be good. It looks, based on the above, like there's a well defined maximum friction braking force requested by the brake mixing algorithm (see around when the pedal was actually pressed at ~252).

willemcvu, 2021-03-02:
Further update - looks like on the other Nissans there's an explicit ```USER_BRAKE_PRESSED``` message.... well I took a look at the Leaf and indeed, we also have this in the CRUISE_THROTTLE message.

Glad I understand how the Leaf braking system works in depth now though :joy: 

I'll test OpenPilot looking at this bit instead of directly at the hydraulic pressure tomorrow night and report back.

vanillagorillaa, 2021-04-06:
@willemcvu Any update on this?

willemcvu, 2021-04-11:
@vanillagorillaa nope, my Leaf has been in the shop for the past 3 weeks, I just got it back. I'll be sure to post an update if I make further progress on this issue.

adeebshihadeh, 2021-05-19:
any progress @willemcvu? would be great to get a fix in for 0.8.5

willemcvu, 2021-05-19:
@adeebshihadeh no progress, sorry. If there's someone else here that has a Leaf that can take up this issue (@razem-io ?) that would be super ideal as I've been too busy at work recently to spend time on this. I haven't even had my C2 installed for the past month :/

razem-io, 2021-05-20:
@willemcvu My time is also very limited at the moment. I plan to get a new fingerprint for my car soon, maybe I find the time to look into this too. As I understand it you already have an updated DBC file? Would you be so kind to provide this file?

willemcvu, 2021-05-20:
Here's my DBC:
https://github.com/willemcvu/opendbc/blob/master/nissan_leaf_2018.dbc

To be honest it isn't in super great shape, it's sort of a first pass at
figuring out a bunch of undocumented messages or messages that are
documented elsewhere.

razem-io, 2021-05-20:
No worries. Thanks! I will try to figure it out.

compbiolover, 2021-08-29:
@razem-io @willemcvu As a fellow 2018 Nissan Leaf owner I was curious if any progress has been made on this bug. Thanks!

razem-io, 2021-08-29:
@compbiolover not yet. Currently working on fingerprinting v2 as this seems to be the bigger issue for now. When this is resolved I will give this a try. Maybe someone else can step in. I guess I can start working on this at the end of this year.

compbiolover, 2021-08-29:
@razem-io No problem. I don't currently have a comma2 or comma3 and my background is in R, but if there is anything I can do once I have a comma device please let me know.

Mr-SourBalls, 2021-09-21:
I am fairly certain @willemcvu had the right idea with the USER_BRAKE_PRESSED message, however i cannot for the life of me get that working in Carstate.py without OP blocking me with an error "waiting for controls to start" 

I added one line to my DBC through SSH under BO_ 569 CRUISE_THROTTLE: 8 XXX
`SG_ USER_BRAKE_PRESSED : 37|1@0+ (1,0) [0|3] "" XXX
`

If i use that DBC to test on a cabana log, it works flawlessly 

If i however dare to touch line 36 of selfdrive/car/nissan/carstate.py from
`ret.brakePressed = bool(cp.vl["BRAKE_PEDAL"]["BRAKE_PEDAL"] > 3)`
to
`ret.brakePressed = bool(cp.vl["CRUISE_THROTTLE"]["USER_BRAKE_PRESSED"] > 0)`

I get the before mentioned error. So it seems the problem lies deeper, or i am missing something.. it LOOKS so straightforward though :(

adeebshihadeh, 2021-11-12:
Fixed thanks to @TheKirkMiller!

closed 2021-11-12 by adeebshihadeh

cellfreak, 2023-01-28:
This is still happening. 2019 Nissan Leaf SL Plus. Comma Three.

TheKirkMiller, 2023-01-28:
What branch/version are you using? I haven't encountered this issue on a 2022 SL Plus/Comma Three since the fix was implemented in the main branch. It might be a different issue you could try to document.

adeebshihadeh, 2023-01-28:
> EDIT: I think I know the issue you're talking about, it's a Nissan Long issue, not related to openpilot. When the lead car completely disappears (like turning off in front of you without merging), with no user input, Nissan Long will automatically disengage, this is stock behavior of the car not controlled by openpilot.

@cellfreak can you confirm it's this issue? if not, I'll reopen this

Mr-SourBalls, 2023-01-28:
Knowing the source of this issue and how it was fixed, it is a different issue. And i suspect Adeeb is right.

i have not had this issue in over 50.000km's