commaai/openpilot #19654: Pre-Collision Malfunction after installing OP on 2021 Highlander hybrid XLE.

fyc2001, 2021-01-05:
**Describe the bug**

car bug: Since @Erich helped me installing OP on 2021 Highlander hybrid XLE. Every time C2 boots up to the camera screen. the "Pre-Collison Malfunction" yellow light and message appears on the dash.  It will stay on unless I turn off the car or unplug C2 while the car is on. 

**How to reproduce or log data**

Turn on the car with C2 plugged in. 

**Expected behavior**

No light. 

**Device/Version information (please complete the following information):**
 - Device: comma two 
 - Dongle ID:  1f8f4de044bdbe9c
 - Route: on all my drives 
 - Timestamp: Every drive after C2 boots up
 - Version: master-ci
 - Car make/model: 2021 Highlander hybrid xle 
 -
**Additional context**

adeebshihadeh, 2021-01-05:
If you'd like to try to fix this yourself, I suspect sending a blank LTA steering message will fix this. Otherwise, we'll try to find some time to look into it before next release.

fyc2001, 2021-01-05:
How do you "send a blank LTA steering message"? I like to give it a try.

adeebshihadeh, 2021-01-05:
Newer Toyotas have an alternate steering interface called LTA. We block the LTA message from the camera to prevent conflicting actuation messages. The fault may be from an ECU expecting the LTA message that we block. To fix this, send a blank (i.e. no torque) LTA message at the same frequency the camera sends it. Here's the code to create an LTA message - https://github.com/commaai/openpilot/blob/master/selfdrive/car/toyota/toyotacan.py#L13.

alancyau, 2021-01-06:
A user with a 2021 Camry Hybrid is also experiencing this issue. 

Dongle ID ffccc77938ddbc44

I see the blank LTA code above, but I'm not sure how to "send" it to the ECU.

adeebshihadeh, 2021-01-12:
Can you test #19731?

alancyau, 2021-01-12:
Is this the proper way to test the new fix? I'm just trying to run the lta branch.

SSH into C2
`mv openpilot mv openpilot.bak`
`git clone -b lta https://github.com/commaai/openpilot && reboot`

adeebshihadeh, 2021-01-12:
After cloning, make sure to also run `git submodule update --init`. We should be able to do this for all the TSS2 cars if this works.

alancyau, 2021-01-12:
Error still shows up. Just tested on pjankovi's 2021 Camry Hybrid. Only thing I added was `CAR.CAMRYH_TSS2`.

fyc2001, 2021-01-15:
Just reporting what I did.
SSH into C2: ran the command. "
mv openpilot mv openpilot.bak  
git clone -b lta https://github.com/commaai/openpilot && reboot"

It said openpilot.bak is not a directory but it continued to the clone part. it ran fine.

after cloning, I ran the command "git submodule update --init" but got a following msg "root@localhost:/data/openpilot$ git submodule update --init
fatal: .git/index: index file smaller than expected"

anyways, I will take C2 home to test it out on the highlander tonight.  Thanks

alancyau, 2021-01-15:
@fyc2001 you need to run that command one directory level up.

SSH into your C2
run `cd /data && rm -rf openpilot && git clone -b lta  https://github.com/commaai/openpilot`
run `git submodule update --init`
run `reboot`

fyc2001, 2021-01-15:
OK, I ran the first line. 
then the second line, I got problem 
"Checking out files: 100% (2254/2254), done.
root@localhost:/data$ **git submodule update --init**
fatal: not a git repository (or any parent up to mount point /)

"

alancyau, 2021-01-15:
> OK, I ran the first line.
> then the second line, I got problem
> "Checking out files: 100% (2254/2254), done.
> root@localhost:/data$ **git submodule update --init**
> fatal: not a git repository (or any parent up to mount point /)
> 
> "

run the submodule command in the openpilot directory. see directions below:

run `cd /data && rm -rf openpilot && git clone -b lta https://github.com/commaai/openpilot`
run `cd openpilot`
run `git submodule update --init`

fyc2001, 2021-01-16:
Ran all three lines. Didn't resolve the LTA malfunction msg and yellow dash light.

WileCyot, 2021-02-02:
I'm having the same issue in my 2021 Camry XSE V6. 

I'm not sure how to do all the things you've suggested above but I'll see if I can't figure it out and try. 

Comma works like a champ but I do get the nice warning message telling me to Visit my Dealer.

pjankovi, 2021-02-13:
Just following up on this to see if anyone has been able figured out a definite solution. I am on the latest branch and it still does the same. Thanks!

fyc2001, 2021-02-13:
From what I searched on discord, WileCyot was able to get rid of the malfunction msg by disabling and enabling openpilot from C2 UI. I tried this on my highlander hybrid 2021 this afternoon, no go. as soon as I enabled openpilot, message came on again.

WileCyot, 2021-02-13:
This was in a 2021 Camry XSE V6 that I performed that on.

I did, however, figured out on my end how to trigger it again.

When I start my car, comma ai works fine. I set it, drive around, all is
well. If I park and shut off the car but then restart it before exiting the
vehicle, boom alarm message like you all see.

I have to put comma into dashcam mode via settings or shut off the car,
open a door (so it shuts down fully), and restart the car.

Weird stuff.

ErichMoraga, 2021-03-17:
After @VirtuallyChris got @EMONAYY6#0998's (DongleID b257184df7026fd4) 2021 Camry FPv2 taken care of (PR https://github.com/commaai/openpilot/pull/20378), we now have another casualty of this bug...
https://discord.com/channels/469524606043160576/524327905937850394/821830482663964673

EMonayy6, 2021-03-17:
The above example is from my vehicle, & once available I am definitely willing to help test any fix that may be available later.

Nyck310, 2021-03-19:
I'm having the same issue as EMONAYY6 (Cruise fault openpilot display, LTA error on dash) on my 21 Camry Nightshade (running 0.8.2.    My DongleID 35a1d2313d50cc5f. If a Bounty starts on this issue I don't mind being included.

edgyman74, 2021-03-29:
I can report I have this same issue for my 2021 Toyota Highlander Platinum edition. Bummer this was pushed to another release 8.4 instead of being part of 8.3.  Think I will return the device in the 30 day window just to be safe in event this issue does not get solved. If it gets addressed I will repurchase. [image]

pjankovi, 2021-03-29:
Did you try to turn off the car, let is sit for 3-4 sec and then turn it on (the car) again? That's what I do and the code is cleared until OP needs another restart. Then I just repeat this process.

edgyman74, 2021-03-29:
> Did you try to turn off the car, let is sit for 3-4 sec and then turn it on (the car) again? That's what I do and the code is cleared until OP needs another restart. Then I just repeat this process. 

Thanks for your reply. So your steps are turn are on car with OP enabled, then turn car off for a few seconds. Then just turn car back on again?  I can try this if my above steps are what your saying.

edgyman74, 2021-04-01:
> Thanks for your reply. So your steps are turn are on car with OP enabled, then turn car off for a few seconds. Then just turn car back on again?  I can try this if my above steps are what your saying.

I can confirm the steps you mentioned work. Start car with OP on, turn off car for a few seconds and back on. Stops complaining for the drive.

pjankovi, 2021-04-01:
> I can confirm the steps you mentioned work. Start car with OP on, turn off car for a few seconds and back on. Stops complaining for the drive.

Yes, sorry I did not get to reply sooner. But, yes, that fixes it. I am glad it worked for you as well!

adeebshihadeh, 2021-04-01:
The PR to fix this (https://github.com/commaai/openpilot/pull/19731) has been updated. Can someone who can reliably reproduce the issue try that PR and let us know if it fixes the issue?

Nyck310, 2021-04-01:
> The PR to fix this (#19731) has been updated. Can someone who can reliably reproduce the issue try that PR and let us know if it fixes the issue?

Just tried the lta branch on my 21 Camry Nightshade, did not fix the issue. Actually its a bit worse because the workaround (starting, remove power, start again) to remove the cruise fault error does not seem to work on this branch. if you have any questions let me know, I'll leave it installed for today in case.

SliderApp, 2021-04-05:
I have the same warning on my 2021 Highlander Platinum ICE. I can trigger it by manually powering down C2, will get message  in dash, when I turn off and turn on the car it will also give me the error while the C2 is powering up.

If I wait for C2 to be fully on, then turn the car on and off it will no longer show the Pre collision  warning.

eFiniLan, 2021-04-06:
I've tried the LTA branch on my unsupported car (2021 C-HR) and also applied the same patch to older versions (0.8.1 / 0.8.2).

I can confirm the patch works.

To me, the 2021 C-HR is a TSS2/2.5 (accurate sensor + full range DRCC) car but both toyota_adas.dbc / toyota_tss2_adas.dbc doesn't parse the radar message correctly, so I have to do more modification to make OP functioning (lateral only for now).

I'm guessing for those ones who still have errors, the issues could be something else?

SliderApp, 2021-04-06:
I can confirm that the LTA branch fixed the error on my 21 Highlander Platinum ICE. I tried several scenarios of starting, turning off, leaving the car, shutting the door and locking it. The only time I could trigger the LTA error was having the car on with engine running, then manually shutting down the C2 via the menu. I would get the error for a few seconds but it would clear and go away by itself.

adeebshihadeh, 2021-04-06:
@eFiniLan glad to here this fixed the issue for you, and would love to get that car upstreamed. The US version of the 2021 CHR is TSS2.5, so looks like we have one case where it's fixed and one where it's not. @Nyck310 are you sure you checked out the branch properly and can you post a route where the fix didn't work? @SliderApp have you tried the fix?

SliderApp, 2021-04-06:
@adeebshihadeh yes it worked for me. Also my highlander has TSS 2.5+, 

pre collision difference from 2.5 : detects oncoming vehicles or pedestrians when performing left-hand turns at intersections, and includes emergency steering assistance to help stabilize the car during emergency maneuvers.

radar cruise control difference from 2.5: TSS 2.5+: When you engage the turn signal while behind a car slower than your preset speed, DRCC will give you a small acceleration boost during and after the lane change to help you get up to your intended speed.

I think that is it.

Nyck310, 2021-04-06:
@adeebshihadeh reinstalled last night. it is showing openpilot 0.8.4 branch lta on the OP HUD while having the same cruise fault as before. Sorry, newer at this, but for posting the route do you want me to attach the Log Segment here?
Thanks for the help so far as well.

Nyck310, 2021-04-06:
@adeebshihadeh My mistake, I think this is the file you were referring to. if you need it in a dif format let me know
[Cruise Fault error.txt](https://github.com/commaai/openpilot/files/6266203/Cruise.Fault.error.txt)

adeebshihadeh, 2021-04-07:
I'll merge the PR since it fixes the issue for at least two models. @Nyck310 did you make sure to update the submodules after pulling the branch?

Nyck310, 2021-04-07:
@adeebshihadeh I followed the instructions above:
 
> run `cd /data && rm -rf openpilot && git clone -b lta https://github.com/commaai/openpilot`
> run `cd /data/openpilot`
> run `git submodule update --init`

adeebshihadeh, 2021-04-07:
I'll close this since it was fixed for the model it was opened for. @Nyck310 can you open a new issue for the Camry? Make sure to fill out the template and also include a short stock route (just disable openpilot before starting the car, Settings -> Toggles -> Enable openpilot).

closed 2021-04-07 by adeebshihadeh