commaai/openpilot #19662: Openpilot unavaiable "communication issue between processes" with ubuntu 20.04 simulator

DanielYEHsieh, 2021-01-05:
**Describe the bug**

followed calar wiki
https://github.com/commaai/openpilot/wiki/CARLA

but gets error message 
Openpilot unavaiable "communication issue between processes"
when attempt to key in "1", "2"

**How to reproduce or log data**

using master branch code

**Expected behavior**

expected to be able to dive like the video
https://www.youtube.com/watch?v=lPQHv3z3MWY

**Additional context**
launch_openpilot.sh log

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: `.' is up to date.
scons: done building targets.
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
getting pilotauth
preimporting selfdrive.thermald.thermald
Building panda firmware
preimporting selfdrive.loggerd.uploader
preimporting selfdrive.loggerd.deleter
preimporting selfdrive.controls.controlsd
preimporting selfdrive.controls.plannerd
preimporting selfdrive.controls.radard
preimporting selfdrive.monitoring.dmonitoringd
preimporting selfdrive.logmessaged
preimporting selfdrive.locationd.locationd
preimporting selfdrive.tombstoned
preimporting selfdrive.pandad
preimporting selfdrive.locationd.calibrationd
preimporting selfdrive.locationd.paramsd
preimporting selfdrive.updated
preimporting selfdrive.rtshield
manager start
{'environ': environ({'SHELL': '/bin/bash', 'NVIDIA_VISIBLE_DEVICES': 'all', 'PYENV_HOOK_PATH': '/root/.pyenv/pyenv.d:/usr/local/etc/pyenv.d:/etc/pyenv.d:/usr/lib/pyenv/hooks:/root/.pyenv/plugins/pyenv-virtualenv/etc/pyenv.d:/root/.pyenv/plugins/pyenv-which-ext/etc/pyenv.d', 'PYTHONUNBUFFERED': '1', 'TMUX': '/tmp//tmux-0/default,8,0', 'HOSTNAME': 'iotlabtpe-Z390-AORUS-PRO-WIFI', 'LANGUAGE': 'en_US:en', 'PYENV_VERSION': '3.8.2', 'PWD': '/openpilot/selfdrive', 'NVIDIA_DRIVER_CAPABILITIES': 'compute,utility,display', 'NOBOARD': '1', 'HOME': '/root', 'LANG': 'en_US.UTF-8', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'PYENV_DIR': '/openpilot/selfdrive', 'SIMULATION': '1', 'PYTHONPATH': '/openpilot:', 'TERM': 'screen', 'TMUX_PANE': '%0', 'DISPLAY': ':1', 'SHLVL': '1', 'PASSIVE': '0', 'PYENV_ROOT': '/root/.pyenv', 'LC_ALL': 'en_US.UTF-8', 'PATH': '/root/.pyenv/versions/3.8.2/bin:/root/.pyenv/libexec:/root/.pyenv/plugins/python-build/bin:/root/.pyenv/plugins/pyenv-virtualenv/bin:/root/.pyenv/plugins/pyenv-update/bin:/root/.pyenv/plugins/pyenv-installer/bin:/root/.pyenv/plugins/pyenv-doctor/bin:/root/.pyenv/plugins/python-build/bin:/root/.pyenv/plugins/pyenv-virtualenv/bin:/root/.pyenv/plugins/pyenv-update/bin:/root/.pyenv/plugins/pyenv-installer/bin:/root/.pyenv/plugins/pyenv-doctor/bin:/root/.pyenv/bin:/root/.pyenv/shims:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin', 'DEBIAN_FRONTEND': 'noninteractive', 'OLDPWD': '/openpilot/tools/sim', 'BASEDIR': '/openpilot', 'DONGLE_ID': '5ea1436c928739d2'})}
selfdrive/loggerd/loggerd.cc: bootlog to /data/media/0/realdata/2021-01-05--08-52-34--0
starting daemon manage_athenad
starting python selfdrive.pandad
starting python selfdrive.thermald.thermald
Connecting to panda
starting python selfdrive.logmessaged
starting process ui
starting python selfdrive.loggerd.uploader
fatal: not a git repository (or any of the parent directories): .git
starting python selfdrive.loggerd.deleter
uploader_fn
{"event": "uploader_netcheck", "is_on_wifi": false}
to upload ('2021-01-05--08-04-18--0/qlog.bz2', '/data/media/0/realdata/2021-01-05--08-04-18--0/qlog.bz2')
{"event": "upload", "key": "2021-01-05--08-04-18--0/qlog.bz2", "fn": "/data/media/0/realdata/2021-01-05--08-04-18--0/qlog.bz2", "sz": 46751744}
checking '2021-01-05--08-04-18--0/qlog.bz2' with size 46751744
uploading '/data/media/0/realdata/2021-01-05--08-04-18--0/qlog.bz2'
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
PulseAudioService: pa_context_connect() failed
PulseAudioService: pa_context_connect() failed
PulseAudioService: pa_context_connect() failed
PulseAudioService: pa_context_connect() failed
[... 177 lines cut ...]

received can message??

received can message??

received can message??

received can message??

[32mpandad[0m [32mthermald[0m [32mlogmessaged[0m [32mui[0m [32muploader[0m [32mdeleter[0m [32mcontrolsd[0m [32mplannerd[0m [32mloggerd[0m [32mradard[0m [32mcalibrationd[0m [32mparamsd[0m [32mcamerad[0m [31mmodeld[0m [32mproclogd[0m [32mlocationd[0m [32mclocksd[0m [31msensord[0m
received can message??

received can message??

received can message??

radarState: Reader was evicted, reconnecting
liveTracks: Reader was evicted, reconnecting
plan: Reader was evicted, reconnecting
pathPlan: Reader was evicted, reconnecting
received can message??

received can message??

Operating system: Ubuntu 20.04

adeebshihadeh, 2021-01-05:
Duplicate of #2501

closed 2021-01-05 by adeebshihadeh

DanielYEHsieh, 2021-01-05:
Hi @adeebshihadeh,

thanks for replying.

I did the same change in
https://user-images.githubusercontent.com/22060625/98455550-2e19d300-2140-11eb-9deb-454dfce0791d.png
but it seems not fixed the issue,

and in bridge.py: is already changed from "greyPanda" to "blackPanda"
but it still runs into the same issue.

kill and restart openpilot did not help too.

Is the fix in discord?

Thanks!