commaai/openpilot #19672: hyundai: palisade has large variation in learned steer ratio

gregjhogan, 2021-01-06:
current learned steer ratio for palisade is definitely too low, but idk what to update it to when it swings this much in a single drive
```
17.8 -> 1ad763dd22ef1a0e|2020-12-29--15-15-43--0
20.7 -> 1ad763dd22ef1a0e|2020-12-29--15-15-43--40
22.3 -> 1ad763dd22ef1a0e|2020-12-29--15-15-43--80
20.6 -> 1ad763dd22ef1a0e|2020-12-29--15-15-43--180
16.5 -> 1ad763dd22ef1a0e|2020-12-29--15-15-43--290
```

gregjhogan, 2021-01-06:
apparently the spec is 15.6, so it is probably variable but haven't found anything that states this is the case
https://www.hyundainews.com/en-us/models/hyundai-palisade-2020-palisade/specifications

zorrobyte, 2021-06-06:
2019 Rav4
[image]
https://www.toyota.com/content/ebrochure/2021/rav4_ebrochure.pdf

Learned ratio:
[image]

I don't think the steer ratio has ever been very spot on, eh?

pd0wm, 2021-06-07:
See my comment in your other PR. I expect a little higher than the factory spec. 

In this case some variability in the learned value between 16.5 and 17 shouldn't be an issue. At a given speed there are many combinations of steer ratio & stiffness that leads to the same handling of the car. The learner needs to look at data at different speeds to find the correct value. Unfortunately the learner has no access to previous data, just the current estimates and current driving data and tries it's best to match that to a steer ratio & stiffness. If you plot stiffness and speed as well you'll probably see a correlation.

I wouldn't be too worried about these small changes. I'm sure at each point in time the stiffness & steer ratio & angle offset estimates matched how the car was driving at that speed. Greg's steer ratio is changing between 16 and 22, which is a much larger variation and is more of a concern.

adeebshihadeh, 2021-11-25:
Is this still an issue after all the locationd/paramsd fixes? @nuwandavek

nuwandavek, 2022-01-05:
We can close this for now. This is not a big deal, but we will look at this in the new paramsd rewrite. There are quite a few things we can change that would circumvent any such filter state issue.

zorrobyte, 2022-01-05:
Always thought it would be nice to have the option to set it statically if known. Many vehicles don’t use variable rate racks and use torque magic in the EPS motor (VW is a notable example of a brand that actually uses it) and if we know what the value should be, why not just use it?

gregjhogan, 2022-01-06:
The learning definitely has nothing to do with variable ratio steering. I believe this value also accounts for more than just the gear ratio of the steering rack (like when the road is wet)?

nuwandavek, 2022-02-14:
Should be fixed with https://github.com/commaai/openpilot/pull/23726

closed 2022-02-14 by nuwandavek