commaai/openpilot #19677: Car battery drain

muttiopenbts, 2021-01-06:
My Honda CR-V 2018 battery was drained when I went out for a drive.
I've noticed that my c2 doesn't switch off when I leave the car.

Someone on general discord channel said the c2 should auto shutdown after 30 hrs or when battery is low.

 - comma two
 - d0a9e4f2bd3fba04
 - d0a9e4f2bd3fba04|2021-01-06--12-04-46
 - Dec 27th around 13:00 PST
 - 0.8.1-release2
 - Honda CR-V 2018

I've also noticed that my c2 frequently goes into unpaired state.

robbederks, 2021-01-11:
Thanks for the report. Dead batteries are always serious.
I made some plots of the metrics in the openpilot logs from your device, from which it seems that the logic was working fine and the device shut off after it estimated the 'reserved' 30Wh was used in your car battery (estimated battery capacity reaching 0, no more logs from the device).

Here are the plots:
[image]
[image]

robbederks, 2021-04-07:
We're working on adding a hardware-in-the-loop CI step to automatically check that all the algorithms we have in place work, and that there is no regressions regarding power draw: #20000. Closing this issue due to inactivity and no bugs found.

closed 2021-04-07 by robbederks