commaai/openpilot #19706: Toyota prius plus (7 seats, 2019):  Car unrecognized

csokaDavidCv, 2021-01-09:
**Describe the bug**
**How to reproduce or log data**

get_fingerprint.py result:
36: 8, 37: 8, 170: 8, 180: 8, 288: 8, 295: 8, 296: 8, 426: 6, 452: 8, 466: 8, 467: 8, 548: 8, 552: 4, 560: 7, 562: 4, 581: 5, 608: 8, 610: 5, 643: 7, 713: 8, 740: 5, 800: 8, 810: 2, 835: 8, 836: 8, 849: 4, 869: 7, 870: 7, 871: 2, 896: 8, 897: 8, 900: 6, 902: 6, 905: 8, 911: 8, 916: 1, 921: 8, 933: 8, 944: 6, 945: 8, 950: 8, 951: 8, 953: 3, 955: 8, 956: 8, 979: 2, 992: 8, 998: 5, 999: 7, 1000: 8, 1001: 8, 1017: 8, 1043: 8, 1044: 8, 1056: 8, 1057: 8, 1059: 1, 1076: 8, 1077: 8, 1114: 8, 1158: 8, 1159: 8, 1161: 8, 1162: 8, 1163: 8, 1184: 8, 1185: 8, 1186: 8, 1190: 8, 1196: 8, 1227: 8, 1232: 8, 1235: 8, 1279: 8, 1422: 8, 1552: 8, 1553: 8, 1554: 8, 1555: 8, 1556: 8, 1557: 8, 1558: 8, 1561: 8, 1562: 8, 1568: 8, 1569: 8, 1570: 8, 1571: 8, 1572: 8, 1584: 8, 1589: 8, 1592: 8, 1593: 8, 1595: 8, 1664: 8, 1728: 8, 1779: 8

fw_versions.py result:

Getting vin...
VIN: JTDZS3EU10J037020
Getting VIN took 0.106 s

  0%|                                                     | 0/3 [00:00<?, ?it/s]                                                              iso-tp query bad response: 0x7f2211
iso-tp query bad response: 0x7f2211
iso-tp query bad response: 0x7f2211
iso-tp query bad response: 0x7f2211
iso-tp query bad response: 0x7f2211
iso-tp query bad response: 0x7f2211
iso-tp query bad response: 0x7f2211
iso-tp query bad response: 0x7f2211
iso-tp query bad response: 0x7f1a12
iso-tp query bad response: 0x7f1a12
iso-tp query bad response: 0x7f0911
iso-tp query bad response: 0x7f0911
iso-tp query bad response: 0x7f0911
iso-tp query bad response: 0x7f3e12
iso-tp query bad response: 0x7f3e12
iso-tp query bad response: 0x7f3e12
iso-tp query bad response: 0x7f3e12
 33%|███████████████                              | 1/3 [00:01<00:02,  1.47s/it]                                                              iso-tp query bad response: 0x7f0911
iso-tp query bad response: 0x7f3e12
 67%|██████████████████████████████               | 2/3 [00:01<00:01,  1.07s/it]                                                              iso-tp query bad response: 0x7f0911
iso-tp query bad response: 0x7f3e12
100%|█████████████████████████████████████████████| 3/3 [00:01<00:00,  1.69it/s]

Found FW versions
{
  (Ecu.esp, 0x7b0, None): [b'F152647280\x00\x00\x00\x00\x00\x00']
  (Ecu.dsu, 0x791, None): [b'881514705100\x00\x00\x00\x00']
  (Ecu.engine, 0x7e0, None): [b'\x0234780000\x00\x00\x00\x00\x00\x00\x00\x00A470                                                              1000\x00\x00\x00\x00\x00\x00\x00\x00']
  (Ecu.fwdRadar, 0x750, 0xf): [b'8821F4702300\x00\x00\x00\x00']
  (Ecu.fwdCamera, 0x750, 0x6d): [b'8646F4703300\x00\x00\x00\x00']
}

Possible matches: set()
Getting fw took 1.824 s

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 4100162388dfc7b0
 - Version: 0.8.1 release
 - Car make/model: [Prius plus (7 seats, 2019) germany]

**Additional context**

vanillagorillaa, 2021-01-10:
Does this car have TSS-P (LKAS and ACC)? I only see ACC in the brochure, no LKAS. You have a camera on the front windshield correct?

csokaDavidCv, 2021-01-10:
The car has a front camera, ACC and LDA + SA (lane departure alert with steering assist). What is the difference between LDA + SA and LKA? As soon as I can, I attach a picture of the vehicle dashboard. What else can I attach to the ticket that will help you?

vanillagorillaa, 2021-01-23:
Please record, and fully upload a new route. Make sure your device is connected to wifi to upload.

vanillagorillaa, 2021-01-25:
You can try this branch/commit to see if this resolves your issue https://github.com/vanillagorillaa/openpilot/commit/29730b0e6bca466a3488094a236f103f50fd2a68

adeebshihadeh, 2021-02-03:
Closing due to inactivity

closed 2021-02-03 by adeebshihadeh