commaai/openpilot #19724: camera_qcom.cc clEnqueueWriteBuffer fails sometimes

pd0wm, 2021-01-11:
Seeing quite a bit of these pop up in sentry.

Tombstone:
```
CM Version: '13.0-20200831-UNOFFICIAL-Sultan-oneplus3'
Build fingerprint: 'OnePlus/OnePlus3/OnePlus3:6.0.1/MMB29M/362280:user/release-keys'
Revision: '0'
ABI: 'arm64'
pid: 6914, tid: 6968, name: processing  >>> ./camerad <<<
signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
Abort message: 'selfdrive/camerad/cameras/camera_qcom.cc:2065: void update_lapmap(MultiCameraState *, const CameraBuf *, const int): assertion "CL_SUCCESS == clEnqueueWriteBuffer(b->q, s->rgb_conv_roi_cl, true, 0, width * height * 3 * sizeof(uint8_t), rgb_roi_buf.get(), 0, 0, 0)" failed'
    x0   0000000000000000  x1   0000000000001b38  x2   0000000000000006  x3   0000000000000000
    x4   0000000000000000  x5   0000000000000001  x6   0000000000000000  x7   0000000000000000
    x8   0000000000000083  x9   21282f1f2b2f1f2b  x10  7f7f7f7f7f7f7f7f  x11  0101010101010101
    x12  0000007f81da5888  x13  7c37eaf726574e1d  x14  7c37eaf726574e1d  x15  00056836e819dabb
    x16  0000007f81d9f560  x17  0000007f81d3116c  x18  0000007f73080d3f  x19  0000007f73081500
    x20  0000007f73081440  x21  0000000000000003  x22  0000000000000006  x23  0000000000000091
    x24  0000007fda87e5f8  x25  0000005562bd7000  x26  0000000000005221  x27  000000000000f663
    x28  0000000000000091  x29  0000007f73080f80  x30  0000007f81d2e908
    sp   0000007f73080f80  pc   0000007f81d31174  pstate 0000000020000000
    v0   28666f657a6973202a2033202a207468  v1   6f725f626772202c29745f38746e6975
    v2   2c65757274202c6c635f696f725f766e  v3   67696568202a206874646977202c3020
    v4   80200802802008000000000000000000  v5   00000000000000000000040000000000
    v6   00000000000000000000000000000400  v7   202c30202c29287465672e6675625f69
    v8   00000000000000000000000000000000  v9   00000000000000000000000000000000
    v10  00000000000000000000000000000000  v11  00000000000000000000000000000000
    v12  00000000000000000000000000000000  v13  00000000000000000000000000000000
    v14  00000000000000000000000000000000  v15  00000000000000000000000000000000
    v16  40100401401004014010040140100401  v17  a00aa00800000004aaa8000400000400
    v18  80200802802008000000000000000400  v19  00000000000000000000000000000000
    v20  00000000000000000000000000000000  v21  ffffffffffffffffffffffffffffffff
    v22  80402010080402018040201008040201  v23  80402010080402018040201008040201
    v24  0000000000000000000000000000fffe  v25  00000000000000000000000000000000
    v26  00000000000000000000000000000000  v27  00000000000000000000000000000000
    v28  00000000000000000000000000000000  v29  00000000000000000000000000000000
    v30  00000000000000000000000000000000  v31  00000000000000000000000000000000
    fpsr 00000010  fpcr 00000000

backtrace:
    #00 pc 000000000006a174  /system/lib64/libc.so (tgkill+8)
    #01 pc 0000000000067904  /system/lib64/libc.so (pthread_kill+68)
    #02 pc 0000000000023848  /system/lib64/libc.so (raise+28)
    #03 pc 000000000001dfe8  /system/lib64/libc.so (abort+60)
    #04 pc 0000000000021430  /system/lib64/libc.so (__libc_fatal+128)
    #05 pc 000000000001e0c0  /system/lib64/libc.so (__assert2+40)
    #06 pc 00000000000116bc  /data/openpilot/selfdrive/camerad/camerad (_Z20camera_process_frameP16MultiCameraStateP11CameraStatei+2396)
    #07 pc 000000000000c5e0  /data/openpilot/selfdrive/camerad/camerad (_Z17processing_threadP16MultiCameraStatePKcP11CameraStatePFvS0_S4_iE+96)
    #08 pc 000000000000d004  /data/openpilot/selfdrive/camerad/camerad (_ZNSt6__ndk114__thread_proxyINS_5tupleIJNS_10unique_ptrINS_15__thread_structENS_14default_deleteIS3_EEEEPFPvP16MultiCameraStatePKcP11CameraStatePFvS9_SD_iEES9_SB_SD_SF_EEEEES7_S7_+52)
    #09 pc 0000000000066d54  /system/lib64/libc.so (_ZL15__pthread_startPv+52)
    #10 pc 000000000001eb94  /system/lib64/libc.so (__start_thread+16)

stack:
         0000007f73080f00  0000000000000000
         0000007f73080f08  0000000000000000
         0000007f73080f10  0000007f73080f60  [stack:6968]
         0000007f73080f18  0000007f81d2e518  /system/lib64/libc.so (_Z23__pthread_internal_findl+80)
         0000007f73080f20  0000007f73081440  [stack:6968]
         0000007f73080f28  0000007f81da7000
         0000007f73080f30  0000000000000003
         0000007f73080f38  0000000000000000
         0000007f73080f40  0000007f73080f60  [stack:6968]
         0000007f73080f48  0000007f81d2e4e4  /system/lib64/libc.so (_Z23__pthread_internal_findl+28)
         0000007f73080f50  0000000000000000
         0000007f73080f58  0000000000000000
         0000007f73080f60  0000007f73080f80  [stack:6968]
         0000007f73080f68  0000007f81d2e8ec  /system/lib64/libc.so (pthread_kill+44)
         0000007f73080f70  0000007f73081500  [stack:6968]
         0000007f73080f78  0000007f73081440  [stack:6968]
    #00  0000007f73080f80  0000007f73080fb0  [stack:6968]
         ........  ........
[... 211 lines cut ...]
         0000007f73081398  0000000000000000
         0000007f730813a0  000000000000000b
         0000007f730813a8  00000000000ff000
         0000007f730813b0  0000007f81e770d8
         0000007f730813b8  0000007f81d9f000  /system/lib64/libc.so
         0000007f730813c0  0000007f730814c0  [stack:6968]
         0000007f730813c8  0000007f78957150  [anon:libc_malloc]
         0000007f730813d0  0000007f73081400  [stack:6968]
         0000007f730813d8  0000005562b95008  /data/openpilot/selfdrive/camerad/camerad (_ZNSt6__ndk114__thread_proxyINS_5tupleIJNS_10unique_ptrINS_15__thread_structENS_14default_deleteIS3_EEEEPFPvP16MultiCameraStatePKcP11CameraStatePFvS9_SD_iEES9_SB_SD_SF_EEEEES7_S7_+56)
    #08  0000007f730813e0  0000007f81d9f000  /system/lib64/libc.so
         0000007f730813e8  0000007f730814c0  [stack:6968]
         0000007f730813f0  0000007f730814c0  [stack:6968]
         0000007f730813f8  0000007f73081440  [stack:6968]
         0000007f73081400  0000007f73081410  [stack:6968]
         0000007f73081408  0000007f81d2dd58  /system/lib64/libc.so (_ZL15__pthread_startPv+56)
    #09  0000007f73081410  0000007f73081430  [stack:6968]
         0000007f73081418  0000007f81ce5b98  /system/lib64/libc.so (__start_thread+20)
         0000007f73081420  0000007f81e77018
         0000007f73081428  0000000000001b02
    #10  0000007f73081430  0000000000000000
         0000007f73081438  0000000000000000
         0000007f73081440  0000007f7327f440
         0000007f73081448  0000007f72f82440
         0000007f'
```

deanlee, 2021-01-12:
maybe related to this : https://github.com/commaai/openpilot/pull/19555

pd0wm, 2021-02-19:
Probably fixed in 0.8.2, need to check the results after it's been in release2 for a bit.

closed 2021-03-03 by pd0wm