commaai/openpilot #19747: Beep when using assistance lane change when it shouldn't.

JustStig, 2021-01-13:
**Describe the bug**
Prior to 0.8.1, when using the assisted lane change, there would be no beeps or alerts, but now in 0.8.1 when using assisted lane change, the comma would beep as if the driver wasn't paying attention during assisted lane changes.

This behavior (beeps) did not happen in 0.8.0 or the prior version.

**How to reproduce or log data**

Using 0.8.1 using assisted lane change at highway speed, the car would beep during lane change.

**Expected behavior**
No beeps or sound made during lane change

**Device/Version information (please complete the following information):**
 - Device: Comma 
 - Dongle ID: 05f2b20ed4718f04
 - Route: Any Route
 - Timestamp: Any time
 - Version: 0.8.1
 - Car make/model: 2020 Hyundai Santa Fe Ultimate

**Additional context**
This did not happen in .8.0 or prior version.

adeebshihadeh, 2021-01-14:
Is the sound from the car or the comma two?

JustStig, 2021-01-15:
The sound from the car and the dash display "keep hands on steering wheel"

adeebshihadeh, 2021-01-22:
Most likely a consequence of #2490. The dash alert is intentional, but the car beeping isn't. @cdwillie76 do you see the same behavior?

cdwillie76, 2021-01-22:
The beep yes. When comma isn’t connected, my Santa Fe beeps also when the keep hands on wheel alert comes up with at stock so I think that is part of the message. Not sure if there is a way to do the hands on wheel without a beep on Santa Fe. Prior to 0.8.1, did the lane change cause the hands on wheel?

JustStig, 2021-01-22:
Prior to 8.1 ALC did not trigger the hands on wheel

Sent from a mobile device.

adeebshihadeh, 2021-02-03:
I understand that. There was a bug fix that shipped in 0.8.1 for the Santa Fe which is why you now see the hands on wheel dash alert. Ideally we want to use the car's dash alerts without the car playing sounds, but it seems the message we currently have for the Santa Fe plays a sound. We'd certainly merge a PR that uses a dash alert that doesn't play a sound if you can find one, but this works as intended.

closed 2021-02-03 by adeebshihadeh