commaai/openpilot #19751: Improving DM Face detection for mask

adam-nelson, 2021-01-14:
I just went thru lockdown for 3 days and mask we're required in private vehicles my state/country

Anytime I have my mask. DM doesnt detect a face and gets unhappy then might disengage unless I nudge the wheel or move my face around

Common Errors:
- Driver distracted
- Asking for wheel nudge. When the driver has hand on the wheel
- no face detected

Discord suggestion was to take the mask off. That just _avoid_ the issue and practical long term for myself or other comma users while covid exist in the world

I have noticed some code on github used similar for improved mask detection
https://github.com/didi/maskdetection
https://www.jumpstartmag.com/didi-open-sourced-the-industrys-first-mask-wearing-ai-detection-technology-with-an-accuracy-rate-of-98/

Device details:
- Device: [comma two]
- Dongle ID: [18e8ac91b805db11]
- Version: [release2 0.8.1]
- Car make/model: [Toyota corolla accent sport hybrid 2019]

geohot, 2021-01-14:
Hmm @ZwX1616 is there mask data in the new dataset?

ZwX1616, 2021-01-30:
@adam-nelson The new model should see some drastic improvement on this. Check out master-ci and let me know if you have any issues!

closed 2021-01-30 by ZwX1616

adam-nelson, 2021-05-14:
After testing OP 0.8.4 - release2

I noticed certain mask type DM get confused on
- Striped/colored mask

However on disposable mask less of a problem