commaai/openpilot #19753: Dangerously late braking

pjankovi, 2021-01-14:
**Describe the bug**

have 2021 Toyota Camry with TSS 2.5+ My main concern is that the car applies brakes too late when approaching stopped car on a hwy. for example there is a stopped car at the red light and I am approaching at 50mph, the car would not start slowing down until almost emergency brakes are needed. 

**How to reproduce or log 

approaching a stopped car on the highway 

**Expected behavior**

It should start reducing speed a lot sooner

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: ffccc77938ddbc44
 - Route: ffccc77938ddbc44%7C2021-01-12--16-31-02
 - Timestamp: 
 - Version: 0.8.2 
 - Car make/model: 2021 Toyota Camry Hybrid LE

abrarmatin, 2021-01-15:
I see the same issue in 2018 Honda Civic. The system reacts as if it does not detect a stopped vehicle in these situations.

pd0wm, 2021-01-15:
You're route is not uploaded completely. 

However this is a known problem. Estimating the speed of stopped cars is a hard problem that we're working on. This is also listed in the list of known limitations: https://github.com/commaai/openpilot#limitations-of-openpilot-acc-and-fcw

pjankovi, 2021-01-15:
Thanks! I appreciate your input!

Best,

Petar Jankovic

hjesse92, 2021-01-16:
I'm interested in how this is solved. Tesla Autopilot seems to have done it quite well in detecting stopped vehicles in front (great for stop and go traffic & local roads :))

pjankovi, 2021-01-17:
The op also works really well in stop and go recognition of a stopped car. The issue is only when approaching a stopped car from far at 50mph or more. But if you approach it with less speed,  whenever starts slowing down, is fine. I think this will get solved when op starts recognizing stop light from far since most of these stopped cars are there because of a red light.

BrettLynch123, 2021-02-03:
Would be good to improve. Very bad with slight curves. I usually disengage slow down until I see on the screen that it detected the lead car and re-engage. But if this issue could be fixed that would be awesome.

isaac-223, 2021-03-25:
Hello, 
I didnt work on this problem yet but looks interesting! @pd0wm If a region (bounding box) of detected car in a lane increases by some ratio with respect to time in a continuous manner, then isnt it a sign of a stopped vehicle? and then the breaks should be applied again with respect to that ratio?

init-22, 2021-03-25:
Hello,
I didnt work on this problem yet but it looks interesting! @pd0wm  If a region (bounding box) of detected car in a lane increases by some ratio with respect to time in a continuous manner, then isnt it a sign of a stopped vehicle? and then the breaks should be applied again with respect to that ratio?

closed 2021-04-20 by pd0wm