commaai/openpilot #19775: Slow start from a dead stop behind a lead car

pjankovi, 2021-01-15:
**Describe the bug**

A faster pull is needed from a dead stop behind a lead car, for  example at the red light. The lead car would be long gone and ahead while mine is very slow catching up. I always need to press on a gas pedal in order to keep up and bot back up the traffic, which then disengages the OP.  is mine start catching up. Toyota’s stock TSS does the same.  I think OP should do this much better. 

**How to reproduce or log data**

Be behind a lead car at the red light light. 

**Expected behavior**

It should start from the stop quicker and use more aggressive gas input in order to be able to follow the lead car. 

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: ffccc77938ddbc44
 - Route: ffccc77938ddbc44|2021-01-14--17-17-02
 - Timestamp: 32min 35sec, or 48min 29sec into the drive
 - Version: 0.8.2
 - Car make/model: 2021 Toyota Camry Hybrid LE

**Additional context**

brianandyt, 2021-01-19:
Have you tried tuning longitudinal?

pjankovi, 2021-01-20:
I have not. I don't know how to do that. Do you have a suggestion?

hjesse92, 2021-01-20:
Same problem (with 2021 Toyota Rav4 Hybrid LE, TSS2.0). I noticed the problem isn't just at a red light, but in general traffic as well. What's odd is that sometimes it performs okay, other times the acceleration is very sluggish to start, then it cranks up very high to catch up.

Wonder if it has something to do with the Hybrid...

pjankovi, 2021-01-20:
Could be, but it could definitely be tuned not to do that. I just don't know how

pjankovi, 2021-01-20:
And yhe fct that you feel sometime it performs well is probably because the car ahead if you has a slow acceleration as well so your car doesn't feel slow. But when the car ahead accelerate at normal speed OP is left in the dust

hjesse92, 2021-01-24:
anyone found a solution for this yet?

pjankovi, 2021-01-24:
I haven't heard any news

ErichMoraga, 2021-01-28:
#tuning in Discord.

pjankovi, 2021-01-29:
Thanks. I cannot find that channel on discord under community.ai. Do I need an invite for #tuning to be able to see?

ErichMoraga, 2021-01-29:
It's under the development section, which you likely collapsed.

BrettLynch123, 2021-02-03:
I have the same issue since V0.8 in a Toyota Corolla Hybrid. Old versions were much better. For me it seems dependent on the car I'm following. Some cars it takes off fine others the lead car will have crossed the entire intersection before mine entered it and it will do it repeatedly. Then sometimes another car cuts in and its fine again..

pjankovi, 2021-03-22:
Update:

I tried so many different tuning options in the attempt to have my 21 Toyota Camry Hybrid follow the lead quicker from the standstill and playing with  parameters for acceleration in planner files, longidutinal, and interface.py in car folder didn’t help. The acceleration gets better, that’s true, but the car still starts the acceleration too late. After experimenting I realized that this may actually be an issue of brake hold vs. electronic brake. I was able to achieve perfect follow, but only after doing the following: 1. Before stopping press brake pedal (disengage OP).2. Use Toyota’s Brake Hold button to keep at standstill behind the lead. 3. Engage OP with “resume” button while on brake hold, and finally 4. Press “resume” once again. After doing all these 4 steps the car would perfectly follow the lead once the lead moves. And I mean really perfect. No delays. It starts accel momentarily. Does anyone have any explanation for this and how can I make OP do the same on its own? Does OP use brake hold or electronic brake at standstill behind the lead? I can hear/feel the bake release, so I am thinking it uses electronic brake and that would explain why it’s so slow in starting the acceleration. Any help is appreciated as I ran out of options. Thanks

hjesse92, 2021-03-26:
Do you engage Brake Hold before step 2, or do you go to complete stop, then engage Brake Hold at step 2?

pjankovi, 2021-03-26:
In Toyota you can click on brake hold at any time prior to this whole procedure, so that brake hold will activate when you come to the full stop and keep your foot on the brake pedal for about 1 sec after full stop. That activates the brake hold. So, you press on the brake hold button (in my car it's behind the gear shift where seat warmer buttons and eco/normal/sport button is) before even step 1. And then when after step 1 you get to a full stop, wait 1 sec for the brake hold to grab the brake. It will light the indicator on your instrument pannel. Aftet it is lit proceed to step 2

TKassis, 2021-05-10:
I just used my Comma Two for the first time today on my Toyota Corolla Hybrid 2021. Very impressive ride, but I also have a few similar issues. 

1- It is very slow to accelerate after a lead car starts moving when fully stopped, but then it accelerates abruptly. Might be an issue with the fact that it is a hybrid? 

2- Related to this, acceleration seems very fast and causes the engine to do the acceleration instead of the electric motor (which is the case when we manually drive hybrids).

3- I also noticed, that when my car is fully stopped on OP the energy visualization on my car shows that power is being supplied from the battery to the motor continuously, this is not the case when I'm stopped without OP engaged (i.e. visualization doesn't show any power being withdrawn from the battery).

Thanks to everyone for all your hard work!

TKassis, 2021-05-10:
After looking at several GitHub issues and other places online, it seems the main problem occurs with the Toyota hybrids. I have no idea how to do the tuning, but will gladly help in any way possible.

alexsielicki, 2021-05-10:
I'm using OP on a 2021 Toyota RAV4 Hybrid and noticed these same issues. Thanks for reporting them @TKassis. Would be nice to do a Toyota Hybrid tune that tries to stay within the "green eco zone" during acceleration by reducing max acceleration but ramping it up quicker from a dead start so there's no need to go into the power zone. I'm also happy to help with that.

As for the energy viz, that's happening to me too. Figured it was just an artifact of Comma being engaged and the system ready to use the battery on start. Not sure how to verify that. Would be strange if it was actually applying electric motor torque against the breaks, but I guess possible.

hjesse92, 2021-08-02:
Has anyone found a solution to this yet? It's still happening to mine on 0.8.6. I thought it either has something to do with the gears' not switching down once at dead stop, or a delay engaging the engine from a dead stop (i.e. OP expects more power from the electric motor but doesn't get it, therefore a delay in acceleration). Just a few thoughts 🤷

TKassis, 2021-08-02:
I stopped using OP except on highways as given it's current capabilities it is safer and less stressful to drive on my own on non-highways/interstates :-). Having said that, when going highway speed OP seems to respond well when there is a stopped car as a result of traffic on the highway. This makes me think it's just a tuning problem. Given the higher car speed at the highway it sends the brake signal a lot sooner (more aggressive). Haven't really spent anytime looking at OP code, so just a theory.

ambientocclusion, 2021-09-06:
I was going to tune for this issue since it's annoying for me too, but seems like it was already improved in master branch.
#21911 Thanks @HaraldSchafer for your work :)
`startingAccelRate` in master branch, which was `startingBrakeRate` before, should be the target value for this issue FYI. startAccel isn't used anywhere yet.

pjankovi, 2021-09-09:
I don't see any difference. I am on 0.8.8 and it still has the same sow start. Does any see any improvement?

pjankovi, 2021-09-14:
Does anybody know what do these values below represent (6.0 and 1.2), and would higher number result in a quicker pull from a standstill? The code is line 332 and 333 from devel branch under selfdrive/car/toyota/interface.py: ret.startingAccelRate = 6.0  # release brakes fast
      ret.startAccel = 1.2  # Accelerate from 0 faster

ambientocclusion, 2021-09-15:
I had some time to test this thing today. Hmm I noticed it's slightly better in 0.8.9 but still not as good as I expected. I checked cabana, and OP sent proper acceleration command immediately when front car is moved, but accelerator indicates quick acceleration was cancelled, then gradually up from low again. Seems like maybe some security stuff exists in behind. Also lead distacne have kinda delay to detect movement too. May need to send customized lead distance commands to tell the car front car is already moved. My car is 2021 Prius Prime (TSS2.0). 

[image]

pjankovi, 2021-09-15:
Thanks for the tests! By the way, do you know what do these values below represent (6.0 and 1.2), and would higher number result in a quicker pull from a standstill? The code is line 332 and 333 from 0.8.9 devel branch under selfdrive/car/toyota/interface.py: ret.startingAccelRate = 6.0  # release brakes fast
      ret.startAccel = 1.2  # Accelerate from 0 faster

Thanks!

Petar Jankovic

ambientocclusion, 2021-09-15:
OP in stock already sent enough accel messages already as you can see above graph ACCEL_CMD, so increasing those parameters don't help for this issue (at least for my car model). Accel messages aren't applied properly to real acceleration due to something else that could be solved by changing other parameters or firmware level security, which is way harder to solve.

ambientocclusion, 2021-09-18:
I think I got little improvement by tuning, but still need to test and organize. Keep posted.

pjankovi, 2021-09-19:
Nice!

ambientocclusion, 2021-10-08:
@pjankovi  Sorry I was busy so took a while to test. My solution is adding the minimum limit of -0.2 for output_accel when car status is standstill. I also changed ACCEL_MAX to 2.0 from 1.5. Still not perfect, but it's better than original behavior on my Prius Prime.
I'm not sure I should make PR since comma is working on e2e_long which might solve this issue already, but you can try my change in my fork repo if you want.
https://github.com/ambientocclusion/openpilot/tree/toyota-stop-go2

pjankovi, 2021-10-08:
Thanks. I will try. In which file did you change adding the minimum limit of -0.2 for output_accel when car status is standstill? 

Best,
P.

ambientocclusion, 2021-10-08:
You can check my commit history in the branch. Openpilot keep throwing close to -2.0 acceleration, which is really strong braking, even car is completely stopped, so I made change to use soft brake at the time.

https://github.com/ambientocclusion/openpilot/commit/dd7d75cf318e37ac0752a59379d01f6f6552554e

adeebshihadeh, 2021-12-15:
A planner bug fixed in 0.8.12 was contributing about a second delay when resuming from a stop. Let us know if you still have issues after updating to 0.8.12.

closed 2021-12-15 by adeebshihadeh

TKassis, 2021-12-15:
Thanks @adeebshihadeh. I'm patiently waiting for the update today before a long drive later tonight :-) I'll test it out tonight. Side note, I did see some improvement for some reason already with 0.8.11. It's still not great when accelerating from a full stop, it still frustrates those behind me, but it is actually more usable than 0.8.10. I'm looking forward to 0.8.12!

pjankovi, 2021-12-20:
Hello! For me 8.12 is still the same in this regard. No improvement as of yet. I absolutely have to add gas manually in order to keep up with the traffic and not to hold the traffic behind me.

BrettLynch123, 2021-12-20:
Same for me

ambientocclusion, 2021-12-20:
It was slow to start for me in master as well with my Prius Prime 2021. What helped this issue was increasing kp and ki tuning values for low speed.
Here are my tuned values worked much better to start acceleration in 0.8.10.

```
kpBP = [0., 10., 20.] 
kpV = [2.0, 1.5, 0.7]
kiBP = [0., 10., 12., 20., 27.]
kiV = [.46, .35, .20, .17, .1]
```

pjankovi, 2021-12-20:
Thanks! Which file are they in?

Petar Jankovic

ambientocclusion, 2021-12-20:
It was in selfdrive/car/toyota/interface.py in past, but seems like it moved to selfdrive/car/toyota/tunes.py in current version. You may need to increase integral values more if you want faster speed than this.

TKassis, 2021-12-20:
Mine is a lot better in 0.8.12. It's actually usable now (without any manual tuning) which wasn't the case before.

HaraldSchafer, 2021-12-21:
The most effective way to make progress on this is to post the tuning layout plots in plotjuggler, with a route ID and of times when it did not accelerate enough

https://github.com/commaai/openpilot/tree/master/tools/plotjuggler#tuning

Changing the PID values can help things, but there are likely better solutions depending on how exactly the plots look.

TKassis, 2022-09-21:
The issue still persists. I now have a new Comma 3 with OP 0.8.16 running on a 2021 Toyota Corolla Hybrid. Acceleration after the lead car starts moving is painfully slow. It seemed to have regressed since using my Comma 2 with OP 0.8.13 (i.e. was behaving a bit better on version 0.8.13)

ecebb0b63434e2f2|2022-09-20--17-41-51--20

thealexbaron, 2022-12-18:
Issue also exists for me running comma 3 OP `0.9.0` in 2020 Hyundai sonata. I love using op on the highway but resuming from a stop almost always requires a disengagement

TKassis, 2022-12-19:
Yeah, it's the same with me on my Corolla on 0.9.0. The experimental e2e mode is even worse.

hjesse92, 2022-12-21:
Same issue with 2021 Rav4 Hybrid. Can we reopen this issue?