commaai/openpilot #19796: Put pending bootlog uploads at the front of the queue

Jon8RFC, 2021-01-17:
They're very small in comparison to everything else.  Regardless of the timestamps, the bootlogs seem to be placed at the end of the queue and behind all routes, in all circumstances.

For situations like sharing a bug report where the bootlogs are necessary, it would be much more convenient to have the tiny bootlogs always upload first instead of uploading gigs of routes for multiple days when the small bootlog from 10 minutes ago is what's wanted.

pd0wm, 2021-01-18:
I'm already working on this! They will also be put in their own folder instead of a separate route.

Jon8RFC, 2021-01-25:
@pd0wm Will the bootlogs be permitted to upload over cellular/comma prime as well?  As far as I can tell, they will only upload over wifi.

pd0wm, 2021-01-25:
Yes. They will be put in a special folder, and uploaded with the qlogs. (Same as crashlogs https://github.com/commaai/openpilot/blob/master/selfdrive/loggerd/uploader.py#L61)

Jon8RFC, 2021-01-26:
Wow, thanks for this! This will be very helpful.

PR #19939 (2021-01-27): Put bootlog in own folder
Fixes #19796

Decided that it was cleaner to just manually open/write the bz2 files than to try integrating this into the other loggerd code. Locking functionality is not needed since this is run by manager before uploader is started.

This PR also removes the sentinel from the bootlog. I don't think those make sense.
closed 2021-01-27 by adeebshihadeh