commaai/openpilot #19801: [Toyota car bug] openpilot can be engaged at zero speed with no lead

cydia2020, 2021-01-17:
**Describe the bug**

openpilot can be engaged at zero speed with no lead by releasing the brakes and pressing the cruise control stalk.

**How to reproduce or log data**

1. stop on hill without lead
2. release brakes (car will not move forward as it's on a hill, so zero speed)
3. engage adaptive cruise
4. openpilot engages and immediately goes into standstill (with comma pedal it presses on the pedal and the car goes out of standstill)

**Expected behavior**

openpilot should display "no close lead" and force acc to disengage

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 399b1cca765e80b4
 - Version: all versions seem to be affected
 - Car make/model: Toyota Prius 2018

**Additional context**

This has been a problem since I firstly started to use openpilot

adeebshihadeh, 2022-06-03:
Sounds like a bug with the standstill. Got a specific route?

cydia2020, 2022-06-03:
No, unfortunately my device can't upload routes to the servers, I can drop a route to a member on discord if that's fine. This is quite easy to reproduce on a TSS-P vehicle though.

sshane, 2022-06-03:
Unfortunately I've seen this on TSS2 as well. On some hills, the car does not respond to our acceleration requests and is unable to get the vehicle moving, not sure if it's related to the car moving backwards.

cydia2020, 2022-06-03:
This is a non issue on stock since the stock system manipulates the lead signal at low speeds, unlike openpilot which sets it to 1 all the time. A very simple solution is that openpilot can just set the lead bit to 0 when the vehicle is stopped (vEgo=0) and there's no lead

cydia2020, 2022-06-03:
The cause of this issue is partially openpilot, but also partially the vehicle. If you send a blank ACC_CONTROL (except MINI_CAR = 1) and engage ACC while stopped without pressing the brakes, the car will also go into resume mode.

cydia2020, 2022-06-03:
We can already cancel cruise when the user is trying to engage without a lead with the brake pressed, so this should be a pretty easy fix

cydia2020, 2022-06-04:
This can be fixed with a one line change in car controller:

```python
lead = hud_control.leadVisible or (CS.out.vEgo < 12. and not (CS.out.vEgo == 0 and not hud_control.leadVisible))
```

sshane, 2023-08-07:
we allow enabling without a lead at a stop, this should be fine to close now. let me know if there's another non-obvious issue here

https://github.com/commaai/openpilot/pull/26632

closed 2023-08-07 by sshane