commaai/openpilot #19810: Make Webcam CI build fast

adeebshihadeh, 2021-01-18:
Our CI job to build openpilot with webcam support is one of our slowest CI jobs. It takes around 12 minutes currently, half of which is just pulling the previously built docker image to use as a cache source. It would be great to get a fully cached webcam build comfortably under 10 minutes.

oliver-ni, 2021-02-06:
We can probably leverage GitHub's cache which might be a lot faster than pulling from the registry every time.

I'll give this a shot this weekend and see if I can get any results.

adeebshihadeh, 2021-02-06:
We already use the cache for routes and our mac CI, so I’m not sure we’ll have enough room. Maybe we can cache just the base image which would make all our jobs faster.

Al-tekreeti, 2021-02-09:
Hello,
can you elaborate more on this issue? I reviewed the job "build_webcam" in "test.yaml". It already uses docker build with cache is enabled. What else needs to be done?
Thanks

Al-tekreeti, 2021-02-09:
I think building the base image is already making use of cache as I can see in the beginning part of test.yaml, or am I wrong?

closed 2021-03-10 by adeebshihadeh