commaai/openpilot #19838: HONDA CR-V 2019 HYBRID  BrakeLights show in CarState:Flags not correct when ACC_CONTROL ON

loveloveses, 2021-01-19:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

my car is HONDA CR-V 2019 HYBRID
I put an lights on car instrument which  get the power supply form my CRV Brake lights 
So when my car  brake lights blink，the light on my car instrument also blink
and i take video check the brake lights blink time compare the signal show in cabana
I fonud my CRV actual Brake lights fellow the rule is：
when acc offset，actual Brake lights is as  [BRAKE_PRESSED][POWERTRAIN_DATA] or [BRAKE_SWITCH][POWERTRAIN_DATA] 
when acc on, actual Brake lights is as [BRAKE_LIGHTS][ACC_CONTROL]

 openpilot use the value [BrakeLights] [CarState:Flags] as [BRAKE_SWITCH][POWERTRAIN_DATA] 
not use [BRAKE_LIGHTS][ACC_CONTROL]

so the actual BrakeLights on my CRV is not as  [BrakeLights] [CarState:Flags] show
when acc offset actual BrakeLights the same as [BrakeLights] [CarState:Flags] show
when acc on，actual BrakeLights the same as [BRAKE_LIGHTS][ACC_CONTROL]
I think it is an bug

**Device/Version information (please complete the following information):**
 - Device: [EON]
 - Dongle ID: [c6d3e563f4ef72bc]
 - Route: [c6d3e563f4ef72bc|2021-01-19--10-14-11]
 - Version: [commit ccdde50d8cd632047cb8c4a7e71659909f4958d7，release2 (0.8.1)]
 - Car make/model: [HONDA CR-V 2019 HYBRID]

I use cabana time and signal check the problem

fellowing is use cabana time，4 period of time in acc on ，actual BrakeLights  the same as [BRAKE_LIGHTS] [ACC_CONTROL] have value

[CONTROL_ON] [ACC_CONTROL]【value 5】
Ftist time.
139-163 second
second time.
336-466 second
third time.
336-466 second
fourth time.
707-757 second

[BRAKE_LIGHTS] [ACC_CONTROL]
Ftist time.
149-152 second
second time.
342-353 second
third time.
452-459 second
fourth time.
6 times actual Brake lights ，easy check on cabana

[BrakeLights] [CarState:Flags] 
which only in fourth time have value
709.41 seond

https://my.comma.ai/cabana/?route=c6d3e563f4ef72bc%7C2021-01-19--10-14-11&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fc6d3e563f4ef72bc%2Fa948d479c3a8175cd030aa108b8c12e1_2021-01-19--10-14-11&seekTime=1

loveloveses, 2021-01-19:
My poor English，i hope can fix it

loveloveses, 2021-01-19:
The problem from selfdrive/car/honda/interface.py
i think it is not car bug，it is openpilot take wrong value in
selfdrive/car/honda/interface.py

line 453
ret.brakeLights = bool(self.CS.brake_switch or
c.actuators.brake > brakelights_threshold)

this is c.actuators.brake have problem

my car is HONDA CR-V 2019 HYBRID

adeebshihadeh, 2021-05-19:
The brakeLights signal was removed in 551091bcfa46e50ef60dac3f3bdff299dba95f35.

closed 2021-05-19 by adeebshihadeh