commaai/openpilot #19865: When changing lanes, it overshoots then corrects (master, 0.8.2)

actuallylemoncurd, 2021-01-21:
**Describe the bug**

When I was on the highway going 65mph, I went to change lanes. After crossing the lane markers i turned my turn signal off (though I've tested this with keeping it on and off for lane change, same affect) OpenPilot proceeded to go almost until the shoulder line, then correct into the center of the lane

**How to reproduce or log data**

Go on highway
Change lanes at around 65, may happen at other speeds but  I was going 65

**Expected behavior**

To change lanes into the center of the other lane, not go to the gutter

**Device/Version information (please complete the following information):**
 - Device: Freon, leeco, grey panda
 - Dongle ID: 579dad05095e1b3e
 - Route: 579dad05095e1b3e|2021-01-21--02-55-24
 - Timestamp: Around 8-9 mins into the drive when i get on the highway
 - Version: https://github.com/actuallylemoncurd/openpilot/tree/0c40b224f4dc85bdec110867b458719c09e34fdb
 - Car make/model: 2013 Volkswagen Jetta Sportwagen TDI

**Additional context**

None afaik

RCoaster, 2021-01-22:
The same thing is happening in 0.8.1, only to the left. Changing lanes to the right works as intended.

actuallylemoncurd, 2021-01-22:
initially i only noticed it going to the right, then i kept testing and it started happening on both

C0mpl3t3N00b, 2021-01-29:
Still happening even on v0.8.2 (latest master-ci from 1/28/2020):

bdad50b8a1f9fa8e|2021-01-28--19-57-49--7

HaraldSchafer, 2021-03-03:
linked to https://github.com/commaai/openpilot/issues/2596

closed 2021-03-03 by HaraldSchafer