commaai/openpilot #19874: Comma two freeze and reboot while engaged

alexklibisz, 2021-01-22:
**Describe the bug**

I recently had an incident on the interstate where my comma two froze completely (while engaged) and rebooted. The video froze, Comma's steering torque turned off, then after about five seconds in this state, the device rebooted.

**How to reproduce or log data**

I don't know. I've driven about 5k miles at this point and this is the first time this has happened. So it seems like quite an anomaly.

**Expected behavior**

Comma should not freeze/reboot while driving :)

**Device/Version information (please complete the following information):**
 - Device: comma2
 - Dongle ID: 75516b36ec7b5bde
 - Route: I don't see any routes on my useradmin?
 - Timestamp: I don't see any routes on my useradmin?
 - Version: v0.8.1-release
 - Car make/model: Toyota Corolla 2017

**Additional context**

I was on a very slight curve on an interstate and had to intervene to prevent going out of lane. I was paying attention, but it took a second to realize what was happening and could have been much worse had it been a sharper curve. Is there any way to have the data/telemetry examined by someone at Comma to make sure there's no obvious defect that caused it? I'm on the latest official release (i.e. no custom fork) and I know the time that it happened. Love the product and appreciate any feedback on this issue.

nabeelr, 2021-01-29:
Almost the exact same issue just happened to me tonight. Slight curve, C2 totally locked up, then rebooted. My only indication was that the path didn't get tighter with the turn, then I got a couple LKA warnings on the dash, before I realized what was going on. It was quite disorienting, and probably the scariest thing my C2 has ever done.

For me, it felt more like 15-30 seconds of being frozen before rebooting.

    Device: comma2
    Dongle ID: 903210ecc0cb706b
    Route: Thu, 28 Jan, 2021 @ 19:43 | 903210ecc0cb706b|2021-01-28--20-37-09--43
    Timestamp: 21:21:30
    Version:  openpilot v0.7.9-VW-release Community Fork
    Car make/model: 2019 VW Golf R Manual

Not sure how to pull logs or if there is a core-dump or panic log I can pull, but more than willing to help out with this however I can.

alexklibisz, 2021-02-12:
Update: Comma support's response is to return/exchange the unit due to presumed hardware failure. It would be nice to know what exactly happened but I get you can't thoroughly investigate every anomaly. Folks at @commma feel free to close this issue.

ketosnacks, 2021-02-25:
Same thing happened to me. Just logging it here in case it's useful to refer to later.

**Description**

Was driving and the the vehicle unexpectedly veered out of the lane during a highway curve (in the overtaking/left lane).

Speed was steady but the lateral control went out, followed by an 'LKA error', then followed by the Comma hanging and rebooting (at this point the car displayed a 'Cruise Control Malfunction' error). When I stopped, turned off the car and rebooted the device and it was fine afterwards.

**How to reproduce or log data**

Unable to reproduce immediately under same external conditions. Data was synced via explorer. There are before and after video segments with a few seconds showing all trace lines disappearing from the OP Comma display prior to veering out of lane.

**Expected behavior**

Maintaining lateral control (and ideally warning of a loss from the device; only the stock warning chimed).

**Device/Version information:**

Device: Comma2
Dongle ID: 74fc73caf41697fa
Route: Wed, Feb 24 @ 06:49 to 10:12
Timestamp: Approximately at 10:11 (with post-device-reboot video in following segments)
Version: OpenPilot v0.8.1
Car make/model: 2021 Toyota Rav4

**Additional context**

My thoughts - it was a bit mucky (snow/slush mist); might have been the radar that threw a code (from being covered in dust). I wiped down the logo where the radar is housed and it didn't seem to happen again. Could also have been the (Toyota LKA) camera. Weather was fair with direct sun from driver's side. Prior to using OP, some highway drives would result in the Toyota throwing an error when the radar/camera would get obstructed by ice/dirt. Could be a possible link.

alexklibisz, 2021-02-25:
Interesting, looks like it's being worked on (copying from linked ticket):

> Also wanted to let you know we are actively working on this. We're looking at it from different angles. We will be increasing the CPU core voltage on devices that are factory tuned to run lower, and we're also looking into any issues caused by the modem driver locking things up.

In my case the incident also followed closely after driving through snowy conditions for a couple hours. But I've driven through plenty of other snowy conditions around New England where comma is just fine 🤷‍♂️ .

nabeelr, 2021-02-25:
So this happened to me one other time since the last time. By far not common, but twice in a month is way more than none in like 6 months of previous ownership.

One thing I was thinking is that I did upgrade NEOS to the latest version, to try OP base, vs the VW fork I'm on right now. When I went back to the VW fork, I did have to downgrade NEOS. Not sure if that is the cause or coincidence.

ketosnacks, 2021-03-07:
Update: Just happened again to me (2nd instance in less than 2 weeks) running the newest version.

Weather conditions were good; driving late night (just before midnight). Windshield and radar/camera were clean. It was a moderate, sustained right curve and I was in the middle lane of a 3 lane highway. The lane markings had some gaps but otherwise it was exactly the same as before; noticed an error on the dash with 'cruise control malfunction' along with loss of steering inputs, then the device hanging for a few seconds and just rebooting. Same follow-up; for the remainder of the drive the Comma device was on and recording but the 'cruise malfunction' would come on any time I re-enabled the cruise control and wouldn't engage. Uploaded the drives.

Route: Sat, Mar 6 @ 23:12 to 23:41
Timestamp: Approximately at 23:41 (new segment starts at 23:42)
Version: OpenPilot v0.8.2

nabeelr, 2021-05-30:
>  I did have to downgrade NEOS. Not sure if that is the cause or coincidence.

Just a much delayed update.
After updating NEOS again, the issue went away for a long while, but now there are other issues that are mentioned in #21057
No clue if they are related at all or not.

ketosnacks, 2021-05-31:
Hasn't reoccured on mine since the latest update, however I do have a suspicion that it might be linked to the automatic sway warning that chimes in on some Toyota, VW, and Hyundai (as far as I know have them) "time for a break?" popups during long periods of cruise control being engaged (usually has a little picture of a coffee cup). Just saw your #21057 mention and thought it might help!

That being said, I did have one instance of the "camera malfunction - contact support" issue which went away after about 2-3 minutes on its own (only thing I did that was noteworthy was driving in direct sunlight).

alexklibisz, 2021-05-31:
Also has not occurred for me since getting a new C2. The original drive where it happened on my previous unit was quite long in total, but I don't remember how long the particular stretch was. I've been on 5 or 6 ~2 hr drives since getting the new unit.

nabeelr, 2021-05-31:
> Hasn't reoccured on mine since the latest update, however I do have a suspicion that it might be linked to the automatic sway warning that chimes in on some Toyota, VW, and Hyundai (as far as I know [...]

Well, for what it's worth, in my ticket #21057, after lots of testing it's looking to be the same issue as described in #20872.

So those two issues might get merged. Thanks for the update!

adeebshihadeh, 2022-01-13:
Tracking this in #21539. A fix should ship in 0.8.13.

closed 2022-01-13 by adeebshihadeh

emersondi, 2022-12-22:
Hi.. my comma2 freeze today... Just start the car... move a few metters and then freeze at 6km/h winth the video frozen too...the través 30 minuts and still frozen.. so I unplugged and plugged the USB... then never turn on... no vibration at plug... and only the blue Led in the back.... what should I do?? Thanks for your suppot and help.