commaai/openpilot #19887: [Kia] eNiro 2019 doesn't show camera fw

VirtuallyChris, 2021-01-23:
User's car matches all existing fw in values.py, but their logs don't display the camera fw so it won't fingerprint. Do we need to add an exception for this car? Is there some fingerprinting bug? User confirmed their car has LKAS and that their harness is hooked up to the camera properly with no dash errors.

4a498ed5529e1e04|2021-01-16--13-21-58

[image]

Jon8RFC, 2021-01-23:
Should I create a bug report for my 2018 Hyundai Sonata as well?  I've been ignored on Discord for a month asking why it's missing.

VirtuallyChris, 2021-01-25:
feel free to add your logs to this bug report too

Jon8RFC, 2021-01-25:
03690f0d837307af|2020-12-29--19-57-21
shows a best case scenario when everything else is present, minus EPS:

transmissionType = unknown,
    carFw = [
      ( ecu = fwdCamera,
        fwVersion = "\xf1\x00LFF LKAS AT USA LHD 1.00 1.01 95740-C1000 E51",
        address = 1988,
        subAddress = 0 ),
      ( ecu = fwdRadar,
        fwVersion = "\xf1\x00LF__ SCC F-CUP      1.00 1.00 96401-C2200         ",
        address = 2000,
        subAddress = 0 ),
      ( ecu = transmission,
        fwVersion = "\xf1\x87\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xf1\x816T6B4051\x00\x00\xf1\x006T6H0_C2\x00\x006T6B4051\x00\x00TLF0G24NL1\xb0\x9f\xee\xf5",
        address = 2017,
        subAddress = 0 ),
      ( ecu = esp,
        fwVersion = "\xf1\x00LF ESC \t 11 \x17\x01\x13 58920-C2610",
        address = 2001,
        subAddress = 0 ),
      ( ecu = engine,
        fwVersion = "\xf1\x81606D5K51\x00\x00\x00\x00\x00\x00\x00\x00",
        address = 2016,
        subAddress = 0 ) ],
    radarTimeStep = 0.05,

koco113, 2021-01-25:
Good evening, Mr. Chris McCammon. What is the way to find the cameras of my
Kia Cadenza k7? There will be no other camera than if the vehicle is placed
in the R position. The camera will operate. I hope that if you have enough
information, this topic will send me a video clip for that. Thank you.

jyoung8607, 2021-06-21:
The camera may be slow to wake up. I had that same problem on VW MQB. We initially had fwdCamera in our FPv2 set, and it would work fine with the car running, or if the C2 had gone into power save shutdown. But if the car was started with openpilot already booted and ready, we would often not get a response from the camera.

openpilot regards fwdCamera as [mandatory if present](https://github.com/commaai/openpilot/blob/28694041d5ee860cdd5f26acb63887e0ce7d8bbf/selfdrive/car/fw_versions.py#L219) in the FPv2 database for that car, so we ended up taking it out of the VW MQB implementation of FPv2 entirely. In our world, the LKAS camera isn't necessarily required for openpilot support, so we needed to deal with cases where it was legitimately absent.

adeebshihadeh, 2021-12-15:
Closing due to inactivity. There have also been improvements to the Hyundai FW query which may have resolved this.

closed 2021-12-15 by adeebshihadeh