commaai/openpilot #19902: HONDA CR-V 2019 HYBRID Can not run master-ci on freon

loveloveses, 2021-01-24:
**Describe the bug**

**Device/Version information (please complete the following information):**
 - Device: [EON Gold]
 - Dongle ID: [1f1e71c163a0750e]
 - Route: [1f1e71c163a0750e|2021-01-24--08-02-12]
 -             [1f1e71c163a0750e|2021-01-24--07-53-36]
 -             [1f1e71c163a0750e|2021-01-24--07-24-54]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [master-ci, Commit  "3bb51ceaa3f45d0fff66d6ed6f5164b7645ccacb"]
 - Car make/model: [HONDA CR-V 2019 HYBRID]

vanillagorillaa, 2021-01-24:
No details are provided on what the actual issue is. Please provide more details.

closed 2021-01-24 by adeebshihadeh

loveloveses, 2021-01-26:
how to show details？