commaai/openpilot #19967: Refactor thneed to include the model pretransform

geohot, 2021-01-29:
Right now, in modeld, we run a bunch of CL kernels to convert to the correct order YUV and warp the image.

These exist outside thneed, but there's no reason they have to. Thneed is a generic OpenCL caching layer, and now it also supports saving and loading from disk. The idea would be to put this so it runs in the thneed compiler.

This can also prevent like 2 useless copies of the image. This is a prereq for bigmodel.

@deanlee Interested?

deanlee, 2021-01-29:
cool, I'm happy to try it out,I need to take some time to understand the code before starting.

geohot, 2021-01-29:
Let me know if you have any questions:

Thneed basically works like this:
1. Run your CL to avoid weird caching
2. Make a Thneed, which is created in record mode
3. Run your CL
4. Disable record mode
5. Now thneed execute will replay your CL exactly with kernel calls, without wasting tons of CPU in the CL stack GPU driver.

deanlee, 2021-01-30:
My first question, after reading thneed's code is: 
the model needs two frames, How to eliminate the memory copies and pass the buffer generated by `frame_prepare` directly to `SNPEModel`  in `thneed`? 
```
float *new_frame_buf = s->frame.frame_prepare(s->q, yuv_cl, width, height, transform);
memmove(&s->input_frames[0], &s->input_frames[MODEL_FRAME_SIZE], sizeof(float)*MODEL_FRAME_SIZE);
memmove(&s->input_frames[MODEL_FRAME_SIZE], new_frame_buf, sizeof(float)*MODEL_FRAME_SIZE);
s->m->execute(&s->input_frames[0], MODEL_FRAME_SIZE*2);
```

geohot, 2021-01-31:
I would have CL keep around the second buffer as an internal buffer and make the copy.

deanlee, 2021-01-31:
Got it

deanlee, 2021-02-03:
I'm working on draft : https://github.com/commaai/openpilot/pull/20005, please correct me in time  if i make any mistakes.

closed 2024-11-07 by adeebshihadeh