commaai/openpilot #19968: Toyota Camry XLE switching from TSS to OP Longitudinal and back

Bradx, 2021-01-29:
Running C2 in a Toyota Camry Hybrid XLE.  Tested Openpilot versions 0.7.10 - 0.8.1.
On fresh boot of the C2, Openpilot runs great, everything stable.  However, if I park the vehicle and turn off the engine for a short stop (to fill up on gas, buy coffee, etc), Once I start the engine (C2 wakes up) the following happens:

1. Openpilot displays an arrow on the screen pointing to the lead car
2. The car subtly accelerates and decelerates in quick cycles.  It sort of feels like I’m towing something. 
3. If lead car slows / stops, Openpilot is slow to see it and slow/stop.  I've had close calls where I've had to slam on brakes.
4. If I force reboot the C2, it resumes normal operation.  Everything is perfect again.
5. Alternatively, if I make a second short stop where I turn off the engine for a few minutes, when I turn the engine back on, the arrow indicating lead car is gone and all is normal

I've tried checking all connections, tried a new usb cable, and tried multiple versions 0.7.10 - 0.8.1.  

Dongle ID: e396297189f42a29
Car Make/Model: Toyota Camry Hybrid XLE 2018

adeebshihadeh, 2021-01-29:
Is your DSU unplugged or do you have a smartDSU installed?

VirtuallyChris, 2021-01-29:
Camry Hybrid (until 2021) is only stock ACC, openpilot can't control gas and brake in that car. This is why you're experiencing 2-3.

Any feelings of different ACC performance is placebo. ( I know how wildly different it can react, had stock ACC in my parents 2018 Rav4 for a while! )

Depending on the version, openpilot would show a lead car indicator when we didn't do ACC but this should be fixed in newer releases. I'd recommend reporting that bug separately if you still see it happen.

adeebshihadeh, 2021-01-29:
Reopening as the logs show that it is switching between openpilot and stock longitudinal control.

Bradx, 2021-01-29:
> 
> 
> Is your DSU unplugged or do you have a smartDSU installed?

I have the DSU which came with my Comma2.  I also had it replaced with a new one as part of a panda failure.  _Commaai sent me replacement parts for everything except the device itself._

adeebshihadeh, 2021-01-29:
Sorry, I missed that this was a Camry, which has its DSU functionality integrated with the radar. The DSU is an ECU inside the car, not comma hardware. Do you have any other hardware (like community hardware) installed besides the comma two and included accessories?

Bradx, 2021-01-29:
Sorry, I got DSU confused with OBD2.   Not sure what community hardware means.  I only have a USB cable plugged into the dash to charge my phone.

adeebshihadeh, 2021-02-20:
Should be fixed by a0a5abc7ad907f1d928f7aac9eba1d8e45154484 which will be included in the 0.8.2 release.

closed 2021-02-20 by adeebshihadeh