commaai/openpilot #20031: V0.8.2 selecting wrong double center lane marking

royboendermaker, 2021-02-06:
**Describe the bug**

I upgraded to the 0.8.2 branch and driving on a 2 way motorway with double center lane markings OP picks the wrong lane line. It picks the outer lane marking that is ment for the opposite lane. So it doesn't position the car on the road in a right way. It's constantly hugging the left lane line. There is aprox. 5-10 cm distance to the lane markings on the left and on the right it's like 50-60 cm. It's undrivable for me this way. With 0.7.9 i had exactly the same problem and will fall back to 0.7.4 that doesn't have this problem and does a decent job.  

**How to reproduce or log data**

Drive on a road with double center lane markings speed around 80 km/h. 

**Expected behavior**

Decent centering of the car on the road with equeal spacing left and right. 

**Device/Version information (please complete the following information):**

    Device: Oneplus 3T, grey panda
    Dongle ID: c4289221cf16a21c
    Route: c4289221cf16a21c|2021-02-06--11-19-45
    Timestamp: Around 12:32;23 till the end, 760s in Cabana.
    Version: https://github.com/actuallylemoncurd/openpilot/tree/pq-legacy-extcan
    Car make/model: 2010 Scirocco 2.0 TDI

**Additional context**
There are 2 people running this branche without this problem, they are based in the US. 
2 people outside of the US and living in the EU have this lane hugging problems.

HaraldSchafer, 2021-04-21:
Laneless policy should deal with this better, if not already it will soon. Once laneless is good enough it will be the default policy.

closed 2021-04-21 by HaraldSchafer

royboendermaker, 2021-07-12:
Laneless didn’t fix it, i tried 0.8.2 laneless and it’s just as bad. Please re-open this bug report.

HaraldSchafer, 2021-07-12:
Laneless was only released in 0.8.3

royboendermaker, 2021-07-13:
Actuallylemoncurd merged the laneless function into 0.8.2 but i understand it’s maybe too far off the Comma code. I will try again with 0.8.5.