commaai/openpilot #20043: Connect to Internet to Engage

charliesexton, 2021-02-09:
**Describe the bug**
Comma 2 has an alert to connect to the internet, an update check is required to Engage. Also has alert,  Connect to internet to check for updates. openpilot won't engage until it connects to the internet to check for updates.
My Comma2 is connected to my home internet and drive logs are up to date.

*

**Device/Version information (please complete the following information):**
 - Device: Comma2
 - Dongle ID: 9363705a7bbefea6
 - Version: V0.8,1
 - Car make/model: 2021 Toyota RAV4

02/08/21 It worked fine this morning but a noon the alerts would display and Comma2 would on engage.

[image]

[image]

adeebshihadeh, 2021-02-09:
You are on a branch that no longer exists. If you uninstall openpilot from the settings and reinstall, you will be on the release branch.

closed 2021-02-09 by adeebshihadeh

charliesexton, 2021-02-09:
I uninstalled and reinstalled and all is good.
Thanks,
Charlie