commaai/openpilot #20044: OpenPilot v0.8.2 will cause wheel to overturn on both directions when at a near standstill behind a stopped lead car.

C0mpl3t3N00b, 2021-02-09:
**Describe the bug**

When following/approaching a stopped lead car, OpenPilot will make large "corrections" from as it is crawling behind the stopped car. In the instance below, my wheel was turned far to the left. When the lead car began moving again, OpenPilot over corrected to the right by making a large turn to the right before correcting again to center.

**How to reproduce or log data**

Follow/approach a lead car that is stopped; road conditions may influence the behavior as this particular instance was not completely straight.

**Expected behavior**

When following/approaching a stopped a lead, my steering wheel should not be making wide turns while at a near stand still.

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: bdad50b8a1f9fa8e
 - Route: bdad50b8a1f9fa8e|2021-02-08--15-55-52--19
 - Timestamp: bdad50b8a1f9fa8e|2021-02-08--15-55-52--19
 - Version: v0.8.2 `master-ci` 003e135ac3fb4e4250adf2dd576a7f022e0290e3
 - Car make/model: 2020 Toyota Rav4 Hybrid LE

**Additional context**

Splinter-nl, 2021-02-09:
I experienced the same thing. Following / approaching a stopped car the steering wheel started making big inputs left and right while at a near stand still. Also the steering wheel became twitchy (shaking) at lower speeds (below 5 mph). 2021 Honda Civic hatch Bosch with modded eps. I was on master-ci 0.8.2 with a custom tune for eps mod.

HaraldSchafer, 2021-02-09:
Should be fixed by this

https://github.com/commaai/openpilot/pull/20053

closed 2021-02-09 by HaraldSchafer