commaai/openpilot #20054: Car will overcompensate towards opposite direction of exit while on freeway

C0mpl3t3N00b, 2021-02-09:
**Describe the bug**

The car will veer away from an exit aggressively almost crossing over the other lane in the direction of the veer.

**How to reproduce or log data**

Drive by an exit at normal highway posted speeds while in the right most lane; this behavior is intermittent but I have experienced it enough to warrant this Bug Report.

**Expected behavior**

Car should stay centered in lane when driving near exit; it should not veer away from exit or toward exit.

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: bdad50b8a1f9fa8e
 - Route: bdad50b8a1f9fa8e|2021-02-08--15-23-32--8
 - Timestamp: bdad50b8a1f9fa8e|2021-02-08--15-23-32--8
 - Version: v0.8.2 `master-ci` 003e135
 - Car make/model: 2020 Toyota Rav4 Hybrid

**Additional context**

[image]

C0mpl3t3N00b, 2021-02-10:
bdad50b8a1f9fa8e|2021-02-09--15-51-06--6

This instance had me go into the other lane before I had to manually steer it back into the appropriate lane.

HaraldSchafer, 2021-02-10:
Lol, I appreciate the clear diagram.

We're working on it!

HaraldSchafer, 2021-03-31:
Improved in 0.8.3

closed 2021-03-31 by HaraldSchafer