commaai/openpilot #20055: Hyundai Ioniq hybrid 2018 enable op but cannot control

andy741217, 2021-02-10:
**Describe the bug**
   When i enable scc,openpilot always will show a green warning "openpilot won't brake while press gas " 
**Device/Version information (please complete the following information):**
 - Device: EON 
 - Dongle ID: 5399c75340359d11
 - Route: 5399c75340359d11|2021-02-09--23-55-55
 - Timestamp: 3min  into the drive
 - Version: release2 0.8.1
 - Car make/model: HYUNDAI IONIQ HYBRID 2018

andy741217, 2021-02-20:
yes, i'm pretty sure i didn't press the gas pedal.

andy741217, 2021-02-20:
one years ago,the same issues.
https://github.com/commaai/openpilot/issues/1757
@adeebshihadeh said The Ioniq Hybrid is a different model,so it will need a new car port.
Can you please add new car port for ioniq hybrid?
coz i wanna use stock op,thx.

pd0wm, 2021-02-22:
In that case we need to figure out why your car is different than the other HKG EVs, and find a bit or signal on CAN to detect when the gas is pressed but that does not trigger when SCC presses the gas.

andy741217, 2021-02-22:
i know many ioniq hybrid owners,we have the same problem.Do i need to provide any information

andy741217, 2021-04-15:
NO Body~~so.................................................???

dorkmo, 2021-04-15:
might need to get on discord and find an expert user to help you guys explore the can messages further

adeebshihadeh, 2021-06-19:
This should be fixed with e7df9adfe3fef777d4ec1143c758ca0501185130. Let us know if you have any issues.

closed 2021-06-19 by adeebshihadeh