commaai/openpilot #20056: KIA SELTOS (2019) OP Engaged but cannot control

wkdwnsghk, 2021-02-10:
OP (EON) IS Engaged but cannot control (=cannot steering)

0.7.3 / 0.7.5 / 0.8.1 / --> all version

https://github.com/wkdwnsghk/OP_ORIGIN/tree/OPKR_0.7.3_HKG_community

OR

https://github.com/wkdwnsghk/OP_NEOKII/tree/neokii75kr

Please Help ~

dorkmo, 2021-02-10:
youre not getting any errors? kinda strange.

I dont know much about the kia seltos, just to double check, does it have a steering assist system built in already? what country are you in? what model year is the car?

the comma might have chosen a fingerprint for a different kia, but perhaps some of the signals are different from your seltos.

you might try to make a fresh fingerprint.

adeebshihadeh, 2021-02-10:
We do not support the Kia Seltos. Open an issue with the fork maintainers.

closed 2021-02-10 by adeebshihadeh